Part-DB/Part-DB-server #385: Create labels based on a category or storage location

jon-daemon, 2023-09-28:
I'd like to be able to create multiple labels at once.
I know that I can create them using their IDs but it would be great to select a category and/or a storage location and create the labels for all the products in this category/storage location.

jbtronics, 2023-09-28:
You can let you display all parts of a category, location etc. using the normal part filter function. Afterwards you select all parts and you can choose to generate labels for all selected parts in the action bar (where you can also select to mass move the parts, delete them, etc.)

jon-daemon, 2023-09-29:
Thank you, I haven't found that.
I think there is a typo in filters for storage location. The last "excluding" should be "including".

[image]

closed 2023-09-29 by jon-daemon