Part-DB/Part-DB-server #389: Unable to login with 2FA

RobThree, 2023-10-03:
**Describe the bug**
I updated to 1.7.2 (master/39009a7) and now I cannot login. Not sure if I messed up the update or what...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `/en/login?_target_path=/login`
2. Enter credentials
3. Click "Login"

**Expected behavior**
I should be logged in

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.7.2 (master/39009a7)
- PHP Version: PHP 8.2.8
- Database Server: 10.11.3-MariaDB-1

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome
 - Version: 117.0.5938.92

**Additional context**
I have restored all permissions to `www-data:www-data` to all files in the project root recursively. I have tried 'disabling' 2FA by (temporarily) setting the `google_authenticator_secret`, `backup_codes` and `backup_codes_generation_date` to `NULL`. This didn't help.  The `prod.log` file contains this:

```
{"message":"Matched route \"2fa_login\".","context":{"route":"2fa_login","route_parameters":{"_route":"2fa_login","_controller":"scheb_two_factor.form_controller::form","_locale":"en"},"request_uri":"https://partdb.mydomain.xxx/en/2fa","method":"GET"},"level":200,"level_name":"INFO","channel":"request","datetime":"2023-10-03T19:42:34.209809+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Checking for authenticator support.","context":{"firewall_name":"main","authenticators":4},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213602+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213638+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213664+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213679+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213696+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213710+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213727+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213740+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.213776+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Read existing security token from the session.","context":{"key":"_security_main","token_class":"Scheb\\TwoFactorBundle\\Security\\Authentication\\Token\\TwoFactorToken"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.216077+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:60 called by App_KernelProdContainer.php:296, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:60 called by App_KernelProdContainer.php:296, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","code":0,"file":"/var/www/partdb/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-10-03T19:42:34.216613+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/var/www/partdb/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:211"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-10-03T19:42:34.222039+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/var/www/partdb/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:211"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-10-03T19:42:34.222500+02:00","extra":{"token":null,"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"User Deprecated: Subscribing to postConnect events is deprecated. Implement a middleware instead. (Connection.php:385 called by Connection.php:450, https://github.com/doctrine/dbal/issues/5784, package doctrine/dbal)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Subscribing to postConnect events is deprecated.  [... cut ...] ifier":"admin"},"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
{"message":"Stored the security token in the session.","context":{"key":"_security_main"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-10-03T19:42:34.242381+02:00","extra":{"token":{"authenticated":true,"roles":[],"user_identifier":"admin"},"url":"/en/2fa","ip":"x.x.x.254","http_method":"GET","server":"partdb.home.mydomain.xxx","referrer":null}}
```

jbtronics, 2023-10-03:
This was caused by some breaking change in the used webauthn library...
It happened when you use Webauthn/U2F as second factor.
This should be fixed in Part-DB 1.7.3 now

FYI: The 2FA with the Webauthn/U2F keys stay active as long as there are keys remaining in the database. So to manually disable all 2FA methods, you have to remove all u2f and webauthn keys from their respective database tables.

closed 2023-10-03 by jbtronics

RobThree, 2023-10-03:
That was fast! Thank you. I updated and it worked first try! Great work. Thank you again!