Part-DB/Part-DB-server #391: Error 500 on Adding a Part with information from a provider

PeterButter, 2023-10-06:
I am getting a 500 error when I try to add a part with information from a provider. I added my API info for Octopart and the search worked but when I try to add the part I get a 500 error and it doesn't work.

[image]

I have installed with the Docket Image and made sure all the permissions are ticked.

[logs.txt](https://github.com/Part-DB/Part-DB-server/files/12830763/logs.txt)

jbtronics, 2023-10-09:
This is most likely caused by some missing permission to create data structures. You should check if your user is allowed to create categories, footprints, manufacturers, suppliers, etc.

However ghere should be a better response in that case instead of the exception.

jbtronics, 2023-10-14:
Now a proper error message is shown below the element selector.

closed 2023-10-14 by jbtronics