Part-DB/Part-DB-server #394: Add "Create new part" button to top bar

RobThree, 2023-10-07:
Would it be possible to add a "Create new part" button in the top bar?

[image]

I am currently in the progress of entering a lot of parts (already did 1246, a few hundred more to go). I keep having to navigate to a random category or to the 'homepage' to get to the "Create new part" button. If it had a bit of a more permanent place (like my suggestion, but anywhere in the top bar will do) it would make my work much easier.

jbtronics, 2023-10-07:
There is now a link on the left side of the navbar. Depending on your permission you can also see a dropdown menu where you can jump to parts import and import from dataprovider: 

[image]

closed 2023-10-07 by jbtronics

RobThree, 2023-10-07:
You. Are. Killing. It! Great work! Thanks!