Part-DB/Part-DB-server #398: Part List doesn't show more than 5 columns and no content for attachments

cooki371, 2023-10-09:
**Describe the bug**
The List of Parts won't show more than five columns defined in the `.env.local`. If I got less than five defined as standard it will just fill the leaving columns with the next column until it reaches 5 columns again.
Additionally the Attachments columns does not contain anything, but the part got a preview image via URL attachment and a pdf file attached to it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any category
2. Take a look at the part list

**Expected behavior**
I expeded the attachments to be shown in the column attachments and all the defined standard columns, not exactly 5.

**Screenshots**
When I got more than 5 defined (`name,description,footprint,storage_location,amount,manufacturer`):
[image]

When I got less than 5 defined(`name,description,store_location`):
[image]

**Server Side**
- Part-DB Version: 1.8.0-dev/commit-5f61e096f9044e3195d8e767b544558216964d22
- PHP Version: 8.1.23
- Database Server: SQLite 3.40.1

**Desktop (please complete the following information):**
 - OS: Windows 11 22H2
 - Browser: Edge Version 117.0.2045.60

jbtronics, 2023-10-09:
The browser saves the last settings of the columns, so that the user can configure which columns he wants to see for himself and override the global settings for his likings (until the browser cache gets deleted).
The env option only affects the initial column visibility, when nothing was stored in the browser yet.

So you need to clear the browser cache (or more exactly Part-DBs local storage in the browser), to see the changes. Or you open Part-DB In a new incognito browser session.

jbtronics, 2023-10-09:
And regarding the attachment column: for historical reasons only attachments with the "show in table" option set.

cooki371, 2023-10-11:
Okay, thank you. Just one question left: Where can I see the name of the column? I want to add the min. stock to the default columns, but can't figure out the name.

jbtronics, 2023-10-11:
`minamount`

See the documentation for that configuration option: https://docs.part-db.de/configuration.html

cooki371, 2023-10-11:
Thanks. Seems like I skipped that part.