Part-DB/Part-DB-server #399: API responsed with error 404 when header contains 'accept: application/json'

cooki371, 2023-10-11:
**Describe the bug**
When I try to call the api on FQDN/api/categories?page=1&itemsPerPage=99 in my app, I get a response with content-type=html/text.
In the beginning when I added accept: application/json to the header, I got a text/html, too. Same with application/jsonld.
Now when I add accept: application/json, I got an http error status 404 with body `{"type":"https:\/\/tools.ietf.org\/html\/rfc2616#section-10","title":"An error occurred","status":404,"detail":"Not Found"}`.
When I am calling the api via the api doc with the same token, I first got a text/html too, but now the proper application/json.

I feel, like I am missing something obvious here.

**To Reproduce**
Steps to reproduce the behavior:
1. Send an http request to the server with the query above.
2. Add accpet: application/json to the header

**Expected behavior**
Response body contains all categories as application/json

**Screenshots**
[image]

**Server Side**
Part-DB Version: 1.8.0-dev/commit-5f61e096f9044e3195d8e767b544558216964d22
PHP Version: 8.1.23
Database Server: SQLite 3.40.1

**Desktop:**
OS: Windows 11 22H2
Browser: Edge Version 117.0.2045.60

**Context**
I am using the http pub package in flutter to send the get request.

jbtronics, 2023-10-14:
If no `Accept` header is set, then the HTML documentation page of the endpoint is set. So when doing API requests you must send either the `Accept` header, or add an `.json` or `.jsonld` suffix to the endpoint to get the desired response type. The correct header type for jsonld is `application/ld+json`.

I can not really replicate the 404 message. I would suspect that your library somehow builds the URL wrongly and end at the wrong endpoint. You should test what happens if you enter `/api/categories.jsonld?page=1&itemsPerPage=99` in your browser (while being logged in)

cooki371, 2023-10-14:
Caught the error with wireshark. The library I used needs the query in a separate Map and doesn't support '?', so the url was just incorrect.

closed 2023-10-14 by cooki371