Part-DB/Part-DB-server #402: Sort Storage locations numerically

sochartgit, 2023-10-11:
**Describe the bug**
Storage locations are sorted in an alphabetically way which is inconsistent when using drawer numbers.

** Actual behavior **
1
10
100
101
.
.
11
110
111

**Expected behavior**
1
2
3
4
5
6
7
8
9
10

jbtronics, 2023-10-15:
The sorting is done by the database, and most databases can order strings only in the binary way not the natural sorting, which you want.
This is a similar problem as issue #243

jbtronics, 2023-10-17:
Yes but that is not portable, as it does not work on MySQL and SQLite. And even then this requires a pretty new MariaDB version.

jbtronics, 2024-06-18:
With the latest commit Part-DB supports natural sort on postgres and MariaDB 10.7+ (and via slow emulation on other platforms)

closed 2024-06-18 by jbtronics