Part-DB/Part-DB-server #411: bin/console cache:clear results in infinite recursive loop

RobThree, 2023-10-16:
**Describe the bug**
`bin/console cache:clear` results in infinite recursive loop

**To Reproduce**
Run `bin/console cache:clear` in the Part-DB root.

**Expected behavior**
A cleared cache

**Screenshots**
[image]

**Server Side**
- Part-DB Version:  1.8.1 (master/cc82547)
- PHP Version: 8.2.8 (SAPI: fpm-fcgi)
- Database Server: 10.11.4-MariaDB-1~deb12u1

**Additional context**
After upgrading from 1.8.0 to 1.8.1 per the instructions in the README, it still showed 1.8.0 so I tried clearing the cache which resulted in this error. Restarting php-fpm/nginx/memcached (not sure if one or all did the trick) resolved the issue.

jbtronics, 2023-10-16:
No there is no infinite loop. It ends at a certain point. But it is deeper than xdebug allows it, by default in older versions.

As far as I am aware that is not a problem in Part-DB, but an effect of the symfony container Generation, which I can not control.

So when you have xdebug enabled (which Is not required in production systems), you need to set `xdebug.max_nesting_level=1000` in your `php.ini` file.

Or if you don't want to change the `php.ini`, you can run `php -d xdebug.max_nesting_level=1000 bin/console cache:clear`

jbtronics, 2023-10-16:
I will try out if that can be set with `ini_set()` directly in the console script.

closed 2023-10-25 by jbtronics