Part-DB/Part-DB-server #414: Error 500 when edit user

iyzpx, 2023-10-18:
**Describe the bug**
When I click on several users in "Tools" -> "Users", an internal server error occurs.
Updated from 1.6.1 (#fa4af99) to 1.8.1 (#cc82547).

Already tried to clear cache sudo -u www-data php bin/console cache:clear and set the owner chown -R www-data:www-data .

log:
[prod.log](https://github.com/Part-DB/Part-DB-server/files/12995239/prod.log)

jbtronics, 2023-10-24:
Have you upgraded the database schema with `php bin/console doctrine:migrations:migrate`?

I can not really replicate that issue, and don't quite understand the reason for the exception in the log. It seems that he somehow tries to access a non-existing user, but I dont know why.

iyzpx, 2023-10-24:
Yep, I've already done that while updating

PartDB Event Log says: `Erfolgreich DoctrineMigrations\Version20230730131708 -> DoctrineMigrations\Version20230816213201`
and when I try it again it also says: `[OK] Already at the latest version ("DoctrineMigrations\Version20230816213201")`.

This does not happen to every user. If I remember correctly, all users where this happens are in the admin group

jbtronics, 2023-10-24:
That was actually quite a tricky bug. This was caused, by the fact that you have deleted the admin user with ID: 2, but that was still referenced in log entries as creator of the users, which failed to display.
Today these references should be deleted, when a user is deleted, however as these log entries were from before the database constraints, they still contained these invalid references. So basically you have some invalid log entries, which caused this strange behavior. I changed some lines in the relevant function and now it seems to correctly return the null value, which means the user does not exists anymore.

closed 2023-10-24 by jbtronics