Part-DB/Part-DB-server #416: Enter date when adjusting stock & ability to adjust dates for stock adjustment

wrightquests, 2023-10-19:
**Is your feature request related to a problem? Please describe.**
I need to be able to adjust the date for when stock is added or removed.

**Describe the solution you'd like**
Ability to enter a date when adjusting stock and/or ability to change the date for stock that has been adjusted.

jbtronics, 2023-10-24:
What date do you want to be able to change? The one put in the log saying when stock was added or removed?

Can you describe your usecase a bit more?

wrightquests, 2023-11-17:
Correct. I'm a solo operation and I catch up on administrative tasks every few weeks. However, I would like my inventory logs to reflect when the part was used.

jbtronics, 2023-11-25:
I'm not a big fan, of rewriting the history that way and let users freely control the "true" timestamp of a log.

However, I have added a field to the modal, where you can set the datetime, when a part was really withdrawn. This is not saved as the timestamp of the log entry (that's still the datetime when the change was made to the DB), but saved and shown as an informational field inside the extra column of the log entry (similar to the comment).

closed 2023-11-25 by jbtronics