Part-DB/Part-DB-server #426: wrong menue design

jojo-re, 2023-10-30:
Hello, 
I've install the Part-DB-server 1.9 dev
with git composer and yarn  on opensuse 15.4 (yes it has php8.2),
in /srv/www/htdocs/Part-DB-server

If  change the APP_ENV=dev  in the .env.local
it seams to be work, but the look is not correct.
I open part-db with the url from my apache2 server. 
No drop-down menus / sidebar and the buttons are only on the left side. 

In the /var/log i found no error
I have forgotten something?

Regards
Juergen

jbtronics, 2023-10-30:
The `APP_ENV=dev` should not be required and you should not have these option unless you want to do development in Part-DB.

Your problem is most likely caused by some missing assets. What errors can you see In the browser console?
Have you run `yarn build` and `php bin/console cache:clear` afterwards?
Do you run part-db on the root of your we host (so on something like `https://part-db.de/` instead of `https://part-db.de/partdb`?

jojo-re, 2023-11-01:
Thanks for your answer :-)
Additional to my first post: I have on the same server an old Part-DB Version 0.6.
It works verry well with an normal / nice theme and design. 

Must I have the virtual host enttry 
or can I open by url  from http://server02/Part-DB-server.1.8/public/   ?

I think the example file for virtual host is to old for the apache version 2.4.51
Have you another example to create a virtual host ?

But the Part-DB-1.9 works without errors in then log file, but nor sidebar no frames in the front end.
I open with the url http://server02/Part-DB-server.1.8/public/de/  on firefox browser.

I found some errors messages:
1,  yarn build:
 WARNING  Be careful when using Encore.configureLoaderRule(), this is a low-level method that can potentially break Encore and Webpack when not used carefully.
62% building 27/28 entries 3850/4000 dependencies 783/975 modules[BABEL] Note: The code generator has deoptimised the styling of /srv/www/htdocs/Part-DB-server.1.8/var/translations/index.js **as it exceeds the max of 500KB**.
99% done plugins FriendlyErrorsWebpackPlugin
After then I send chown -R wwwrun:www /srv/www/htdocs/Part-DB-server.1.8/ 
and it works withot sidebar and frames.
If I press the buttons, no response.
[image]
[image]
[image]

Please see the attached  screeshot.

2.  The package nodejs:
I have installed nodejs  one my opensuse 15.4 system with yast.
more versions  node, node-default, node12, node14, node16, node18
note-default is version v18.18.2

3. If Part-DB started:
Part-DB-1.9
Version: 1.9.0-dev (master/1409d19)
Lade Markdown. Wenn diese längere Zeit bestehen bleibt, versuchen sie die Website erneut zu laden!
Lizenzinformation
How can I check if markdown was load?
On my Linux (opensuse 15.4 system it is installed.

jbtronics, 2023-11-01:
> Must I have the virtual host enttry
> or can I open by url from http://server02/Part-DB-server.1.8/public/ ?

It is highly recommended to set the DocumentRoot to the public folder of Part-DB (by configuring a seperate vhost). You get cleaner URLs and without a proper configuration, any user can access all (secret) files in the Part-DB folder. 
If you really want to put Part-DB in a subdirectory (and use the `/public/` folder), be sure to forbid access to the folders above the `public/` folder. Also you will need to change some things in the `webpack.config.js` file to configure the path prefix:

> If you have put Part-DB into a subdirectory on your server (like part-db/), you have to edit the file webpack.config.js and uncomment the lines (remove the // before the lines) .setPublicPath('/part-db/build') ( line 43) and .setManifestKeyPrefix('build/') (line 44). You have to replace /part-db with your own path on line 44.

Be sure to do a new `yarn build` and `php bin/console cache:clear` afterwards.

jbtronics, 2023-11-03:
Just for documentation purposes: In the upcoming Part-DB 1.9 version, changing the publicPath in `webpack.config.js` will not be needed anymore, instead the path is detected automatically.

jojo-re, 2023-11-06:
Hi, after the last update it works fine.
Thank you for working on this projekt.

closed 2023-11-08 by jbtronics