Part-DB/Part-DB-server #427: Error 500 behind Nginx Proxy Manager

burgsth, 2023-11-01:
I try to run Part-DB in a docker container behind a "Nginx Proxy Manager" reverse proxy. 
I've added DEFAULT_URI and TRUSTED_PROXIES to my docker compose but everything I get is a "500
Internal Server Error" generated by Part-DB.

My docker compose looks like this:

```
version: '3.3'
services:
  partdb:
    container_name: partdb
    # By default Part-DB will be running under Port 8080, you can change it here
#    ports:
#      - '8080:80'
    volumes:
      # By default
      - /mnt/user/appdata/partdb/uploads:/var/www/html/uploads
      - /mnt/user/appdata/partdb/public_media:/var/www/html/public/media
      - /mnt/user/appdata/partdb/db:/var/www/html/var/db
    restart: unless-stopped
    image: jbtronics/part-db1:latest
    environment:
      # Put SQLite database in our mapped folder. You can configure some other kind of database here too.
      - DATABASE_URL=sqlite:///%kernel.project_dir%/var/db/app.db
      # In docker env logs will be redirected to stderr
      - APP_ENV=docker

      # You can configure Part-DB using environment variables
      # Below you can find the most essential ones predefined
      # However you can add add any other environment configuration you want here
      # See .env file for all available options or https://docs.part-db.de/configuration.html

      # The language to use serverwide as default (en, de, ru, etc.)
      - DEFAULT_LANG=de
      # The default timezone to use serverwide (e.g. Europe/Berlin)
      - DEFAULT_TIMEZONE=Europe/Berlin
      # The currency that is used inside the DB (and is assumed when no currency is set). This can not be changed later, so be sure to set it the currency used in your country
      - BASE_CURRENCY=EUR
      # The name of this installation. This will be shown as title in the browser and in the header of the website
      - INSTANCE_NAME=Part-DB

      # Allow users to download attachments to the server by providing an URL
      # This could be a potential security issue, as the user can retrieve any file the server has access to (via internet)
      - ALLOW_ATTACHMENT_DOWNLOADS=0
      # Use gravatars for user avatars, when user has no own avatar defined
      - USE_GRAVATAR=0

      # Override value if you want to show to show a given text on homepage.
      # When this is empty the content of config/banner.md is used as banner
      #- BANNER=This is a test banner<br>with a line break
	  # The public reachable URL of this Part-DB installation. This is used for generating links to the website in emails and so on
      # This must end with a slash!
      - DEFAULT_URI="https://partdb.lan/"
      
      # Set the trusted IPs here, when using an reverse proxy
      #TRUSTED_PROXIES=127.0.0.0/8,10.0.0.0/8,172.16.0.0/12,192.168.0.0/16
      - TRUSTED_PROXIES=172.23.0.0/16,192.168.1.10  
  
networks:
  default:
    external: true
    name: proxy
```

Container log when I try to access the page looks like this

```
[01-Nov-2023 21:21:02] NOTICE: fpm is running, pid 26
[01-Nov-2023 21:21:02] NOTICE: ready to handle connections
[01-Nov-2023 21:21:02] NOTICE: systemd monitor interval set to 10000ms
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.23.0.10. Set the 'ServerName' directive globally to suppress this message
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.23.0.10. Set the 'ServerName' directive globally to suppress this message
[Wed Nov 01 21:21:02.983859 2023] [mpm_event:notice] [pid 1:tid 23293517544768] AH00489: Apache/2.4.56 (Debian) configured -- resuming normal operations
[Wed Nov 01 21:21:02.983935 2023] [core:notice] [pid 1:tid 23293517544768] AH00094: Command line: 'apache2 -D FOREGROUND'
200 Matched route "redirector". | context={"route":"redirector","route_parameters":{"_route":"redirector","_controller":"App\\Controller\\RedirectController::addLocalePart","url":""},"request_uri":"http://partdb.lan/","method":"GET"} level_name=INFO channel=request datetime=2023-11-01T21:21:31.609398+00:00 extra={"token":null,"url":"/","ip":"192.168.1.228","http_method":"GET","server":"partdb.lan","referrer":null}
{"message":"Uncaught PHP Exception OneLogin\\Saml2\\Error: \"Invalid array settings: sp_acs_url_invalid, sp_sls_url_invalid\" at /var/www/html/vendor/onelogin/php-saml/src/Saml2/Settings.php line 141","context":{"exception":{"class":"OneLogin\\Saml2\\Error","message":"Invalid array settings: sp_acs_url_invalid, sp_sls_url_invalid","code":2,"file":"/var/www/html/vendor/onelogin/php-saml/src/Saml2/Settings.php:141","trace":["/var/www/html/vendor/onelogin/php-saml/src/Saml2/Auth.php:178","/var/www/html/vendor/nbgrp/onelogin-saml-bundle/src/Onelogin/AuthFactory.php:25","/var/www/html/var/cache/docker/ContainerDxnJ7En/getAuthRegistryInterfaceService.php:22","/var/www/html/var/cache/docker/ContainerDxnJ7En/App_KernelDockerContainer.php:228","/var/www/html/var/cache/docker/ContainerDxnJ7En/getSecurity_Authenticator_Saml_MainService.php:38","/var/www/html/var/cache/docker/ContainerDxnJ7En/App_KernelDockerContainer.php:228","/var/www/html/var/cache/docker/ContainerDxnJ7En/getSecurity_Authenticator_Manager_MainService
.php:30","/var/www/html/var/cache/docker/ContainerDxnJ7En/App_KernelDockerContainer.php:228","/var/www/html/var/cache/docker/ContainerDxnJ7En/getSecurity_Firewall_Authenticator_MainService.php:20","/var/www/html/var/cache/docker/ContainerDxnJ7En/App_KernelDockerContainer.php:228","/var/www/html/var/cache/docker/ContainerDxnJ7En/getSecurity_Firewall_Map_Context_MainService.php:40","/var/www/html/vendor/symfony/security-bundle/Security/LazyFirewallContext.php:48","/var/www/html/vendor/symfony/security-http/Firewall.php:128","/var/www/html/vendor/symfony/security-http/Firewall.php:95","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:260","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:220","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:56","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:157","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76","/var/www/html/vendor/symfony/http-kernel/Kernel.php:197","/var/www/html/vendor/symfony/runtime/
Runner/Symfony/HttpKernelRunner.php:35","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/public/index.php:5"]}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2023-11-01T21:21:31.615195+00:00","extra":{"token":null,"url":"/","ip":"192.168.1.228","http_method":"GET","server":"partdb.lan","referrer":null}}
- -  01/Nov/2023:21:21:28 +0000 "GET /index.php" 500
172.23.0.6 - - [01/Nov/2023:21:21:28 +0000] "GET / HTTP/1.1" 500 4220 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/118.0.0.0 Safari/537.36"
```

IP of NPM reverse proxy is 172.23.0.6/16, PartDB has got 172.23.0.10/16. My docker host listens to 192.168.1.10.
I've also tried different  TRUSTED_PROXIES variants without any different result (e.g. set to 172.23.0.6).

jbtronics, 2023-11-01:
I think that is related to the `DEFAULT_URI` env setting. I will look into it.

As workaround you can try to just comment the DEFAULT_URI env setting or set it to an empty value. This value is mostly only needed for generating links in emails. When using Part-DB via browser it should work fine without that.

burgsth, 2023-11-01:
Removing `DEFAULT_URI` fixes the issue

jbtronics, 2023-11-11:
I am not really able to replicate this exception.
What version of Part-DB were you using?

burgsth, 2023-11-13:
The image I'm using ist

jbtronics/part-db1:latest
sha256:708bc2ff0c517c03b8c5f8cc89dda4bc7459128b6b279c63d3f5814f28d8a8b5 
Docker on linux, amd64

Here is my docker compose
`version: '3.3'
services:
  partdb:
    container_name: partdb
    # By default Part-DB will be running under Port 8080, you can change it here
#    ports:
#      - '8080:80'
    volumes:
      # By default
      - /mnt/user/appdata/partdb/uploads:/var/www/html/uploads
      - /mnt/user/appdata/partdb/public_media:/var/www/html/public/media
      - /mnt/user/appdata/partdb/db:/var/www/html/var/db
    restart: unless-stopped
    image: jbtronics/part-db1:latest
    environment:
      # Put SQLite database in our mapped folder. You can configure some other kind of database here too.
      - DATABASE_URL=sqlite:///%kernel.project_dir%/var/db/app.db
      # In docker env logs will be redirected to stderr
      - APP_ENV=docker

      # You can configure Part-DB using environment variables
      # Below you can find the most essential ones predefined
      # However you can add add any other environment configuration you want here
      # See .env file for all available options or https://docs.part-db.de/configuration.html

      # The language to use serverwide as default (en, de, ru, etc.)
      - DEFAULT_LANG=de
      # The default timezone to use serverwide (e.g. Europe/Berlin)
      - DEFAULT_TIMEZONE=Europe/Berlin
      # The currency that is used inside the DB (and is assumed when no currency is set). This can not be changed later, so be sure to set it the currency used in your country
      - BASE_CURRENCY=EUR
      # The name of this installation. This will be shown as title in the browser and in the header of the website
      - INSTANCE_NAME=Part-DB

      # Allow users to download attachments to the server by providing an URL
      # This could be a potential security issue, as the user can retrieve any file the server has access to (via internet)
      - ALLOW_ATTACHMENT_DOWNLOADS=0
      # Use gravatars for user avatars, when user has no own avatar defined
      - USE_GRAVATAR=0

      # Override value if you want to show to show a given text on homepage.
      # When this is empty the content of config/banner.md is used as banner
      #- BANNER=This is a test banner<br>with a line break
	  # The public reachable URL of this Part-DB installation. This is used for generating links to the website in emails and so on
      # This must end with a slash!
      - DEFAULT_URI="http://partdb.lan/"
      
      # Set the trusted IPs here, when using an reverse proxy
      #TRUSTED_PROXIES=127.0.0.0/8,10.0.0.0/8,172.16.0.0/12,192.168.0.0/16
      - TRUSTED_PROXIES=172.23.0.0/16,192.168.1.10  
  
networks:
  default:
    external: true
    name: proxy`

Behind a NGINX Proxy Manager docker instance.
Same behavoure on my vServer with wildcard SSL Certificate

burgsth, 2023-11-14:
Removing the quotes fixed it, thank you!

closed 2023-11-19 by jbtronics