Part-DB/Part-DB-server #429: Add stock column to project bom list

cooki371, 2023-11-06:
**Is your feature request related to a problem? Please describe.**
Currently the BOM list of a project doesn't show the parts location. You have to click on the part and navigate to the stock to show the current location.

**Describe the solution you'd like**
I'd like to add a column to the BOM list which shows the current location(s) of the part.
Same when in build areas BOM list.

**Additional context**
The BOM list isn't very useful when the stock location isn't shown.

ECCOsea, 2023-11-10:
+1 very good idea!

jbtronics, 2023-11-19:
There are now columns showing the stocked amount and storage locations of the linked Part in the BOM.
They are hidden by default, but can be added by the user easily via the column visibility selector.

closed 2023-11-19 by jbtronics