Part-DB/Part-DB-server #431: Mouser not working despite env correctly filled

sochartgit, 2023-11-13:
Part-DB 1.8.2 running on Docker without any problem.
Env variable setup:
      - PROVIDER_MOUSER_KEY=x1x1x1x1x1x11x1x1x1x1

Mouser keeps being disabled with error message:

 This provider uses the Mouser API to search for parts.
Basic Picture Datasheets Prices
Disabled
Configure the API key in the PROVIDER_MOUSER_KEY environment variable to enable.

sochartgit, 2023-11-13:
And yes, I've checked several times, and yes that's the Search key provided by Mouser.
The variable is accessible to the container by  an:    echo $PROVIDER_MOUSER_KEY

jbtronics, 2023-11-14:
Have you cleared the cache after setting the secret?

sochartgit, 2023-11-14:
I do and I get an error 500 now when I go to  /en/tools/info_providers/search
[image]

sochartgit, 2023-11-14:
And I get this in the  docker logs:

{"message":"Failed to save key \"version\" of type Version\\Version: fopen(/var/www/html/var/cache/docker/pools/system/pbl-hE1W7B/F28wq-1u): Failed to open stream: Permission denied","context":{"key":"version","exception":{"class":"ErrorException","message":"fopen(/var/www/html/var/cache/docker/pools/system/pbl-hE1W7B/F28wq-1u): Failed to open stream: Permission denied","code":0,"file":"/var/www/html/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:95","trace":["/var/www/html/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:95","/var/www/html/vendor/symfony/cache/Adapter/PhpFilesAdapter.php:248","/var/www/html/vendor/symfony/cache/Adapter/AbstractAdapter.php:148","/var/www/html/vendor/symfony/cache/Traits/AbstractAdapterTrait.php:239","/var/www/html/vendor/shivas/versioning-bundle/src/Service/VersionManager.php:116","/var/www/html/vendor/shivas/versioning-bundle/src/Twig/VersionExtension.php:31","/var/www/html/vendor/twig/twig/src/ExtensionSet.php:331","/var/www/html/vendor/twig/twig/src/Environmen

sochartgit, 2023-11-14:
Solved error 500 by applying chmod to the /var/www/html folder but Mouser is still Disabled.

jbtronics, 2023-11-19:
> {"message":"Failed to save key "version" of type Version\Version: fopen(/var/www/html/var/cache/docker/pools/system/pbl-hE1W7B/F28wq-1u): Failed to open stream: Permission [...]

That is just an unrelated issues, caused because you have cleared the cache as the wrong user (you must do it as the web server user, or the permissions get wrong).

I can not replicate the main issue, and I do not see a cause for it. The provider should be active, as soon as the mouser key env is not empty.
Have you checked that there is no additional configuration, which could overwrite the settings? Like a `.env.local` file or some modified `config/services.yaml` file inside the docker container?

sochartgit, 2023-11-20:
Yes I do, no .env.local and I get the right env. variable from an echo in my container.
So the variable is correctly provided. And the key is the right one too.

sochartgit, 2023-11-23:
[image]

jbtronics, 2023-11-25:
I have tried it myself on a docker container, and mouser gets enabled just fine, like expected, when adding that line to the docker-compose file.

Can you send the complete `docker-compose.yaml` file?

sochartgit, 2023-11-25:
version: '3.3'
services:
  partdb:
    container_name: partdb
    ports:
      - 192.168.10.8:443:443
    volumes:
      - uploads:/var/www/html/uploads
      - public_media:/var/www/html/public/media
      - db:/var/www/html/var/db
      - apache:/etc/apache2
      - /root/.acme.sh:/certs
    restart: unless-stopped
    image: jbtronics/part-db1:latest
    networks:
      web:
    environment:
      # Put SQLite database in our mapped folder. You can configure some other kind of database here too.
      - DATABASE_URL=sqlite:///%kernel.project_dir%/var/db/app.db
      # In docker env logs will be redirected to stderr
      - APP_ENV=docker
      # You can configure Part-DB using environment variables
      # Below you can find the most essential ones predefined
      # However you can add add any other environment configuration you want here
      # See .env file for all available options or https://docs.part-db.de/configuration.html

      - MAILER_DSN=smtp://smtp.XXXX.ca:25

      - EMAIL_SENDER_EMAIL=partdb_alert@XXXX.ca #The email address from which emails should be sent from (in most cases this has to be the same as the email address used for SMTP access)
      - EMAIL_SENDER_NAME=Part-DB #Similar to EMAIL_SENDER_EMAIL but this allows you to specify the name from which the mails are sent from.
      - ALLOW_EMAIL_PW_RESET=true
      - BANNER= "T'es organisé l'gros"
      # The language to use serverwide as default (en, de, ru, etc.)
      - DEFAULT_LANG=en
      # The default timezone to use serverwide (e.g. Europe/Berlin)
      - DEFAULT_TIMEZONE=America/Toronto
      # The currency that is used inside the DB (and is assumed when no currency is set). This can not be changed later, so be sure to set it the currency used in your country
      - BASE_CURRENCY=CAD
      # The name of this installation. This will be shown as title in the browser and in the header of the website
      - INSTANCE_NAME=PARTS

      # Allow users to download attachments to the server by providing an URL
      # This could be a potential security issue, as the user can retrieve any file the server has access to (via internet)
      - ALLOW_ATTACHMENT_DOWNLOADS=1
      # Use gravatars for user avatars, when user has no own avatar defined
      - USE_GRAVATAR=0

      # Override value if you want to show to show a given text on homepage.
      # When this is empty the content of config/banner.md is used as banner
      #- BANNER=This is a test banner<br>with a line break
     # - SAML_ENABLED=true

      - PROVIDER_MOUSER_KEY=AAAAAAA-BBBB-AAAA-YYYY-XXXXXXXXXX

      - PROVIDER_DIGIKEY_CLIENT_ID=ABABABABABABABABABABABA
      - PROVIDER_DIGIKEY_SECRET=XXXXXXXXXXXX
      - PROVIDER_DIGIKEY_CURRENCY=CAD
      - PROVIDER_DIGIKEY_COUNTRY=CA

      - PROVIDER_OCTOPART_CLIENT_ID=XXXXXXXXX-YYYY-ZZZZ     
      - PROVIDER_OCTOPART_SECRET=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
      - PROVIDER_OCTOPART_CURRENCY=CAD
          
      - PROVIDER_TME_KEY=XXXXXXXXXXXXXXXXXXXXXXXXXX
      - PROVIDER_TME_SECRET=XXXXXXXXXXXXXXXXXX

      - PROVIDER_ELEMENT14_KEY=XXXXXXXXXXXXXXXXXXXXXXXX

    labels:
      - com.centurylinklabs.watchtower.enable=true  

networks:
  web:
    external: true
volumes:
  uploads:
  public_media:
  db:
  apache:

sochartgit, 2023-11-27:
Trouble solved. The problem was not the SSL itself as I've done that hundreds of times.
Problem was symfony.conf being persistent and not upgraded to latest one when you added Mouser as provider. So every other provider was working fine because the PassEnv were filled.
I've added the right line to symfony.conf and now it works.

closed 2023-12-07 by jbtronics