Part-DB/Part-DB-server #434: HTTP 500 errors occur when logging in, prevent parts from loading, trying to edit

virtadpt, 2023-11-19:
**Describe the bug**
I've just migrated my existing install to a new server for performance's sake and now I'm getting HTTP error 500's when I log in, when I try to load a part that is already in the database, or whenever I try to edit something.  This happens in both a regular and a Private Browsing window.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to my Part-DB-server install.
2. Enter credentials and click "Login."
3. Get "Login successful" modal.
4. Get "Internal Server Error (Status 500) - Error calling /en/."  There is nothing useful in the "View details" part, only "500 Internal Server Error"
5. Scroll to the bottom of the screen.  The "Last activity" part says "Loading - This can take a moment. If this message do not disappear, try to reload the page" and never goes away.
6. Click on any category.  It loads.
7. Click on any part in the category.  Get a "Internal Server Error (Status 500) - Error calling https://merivis.virtadpt.net/en/part/100/info. Try to reload the page or contact the administrator if this error persists." modal.  There is nothing useful in the "View details" part, only "500 Internal Server Error"
8. Click "OK."  The part isn't shown, I just see the category as before.
9. Tools -> Edit -> Pick one of the things (let's say "Storage locations").  Click on a storage location.  Get a full screen (well, browser window) "500 - Internal Server Error".
10. Tools -> System -> Users.  Click on a user.  Get a full screen (well, browser window) "500 - Internal Server Error".

**Expected behavior**
After logging in, I see the last couple of edits I've made in my inventory.

When clicking on an entry in a category, I see that entry and have the option of editing it.

When clicking on something specific in the Tools menu (Tools, Edit, Show, System), I don't get a 500 error.

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.9.0-dev (commit 3da656c08b21dfc03d280f0e6d9ea6e4f4cac0ec)
- PHP Version: v8.1.24
- Database Server MySQL 8.0.35

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox
 - Version: 117.0.1 (64-bit)

**Additional context**
In the var/log/prod.db file I see the following (reformatted a little to minimize horizontal scrolling):

```
{"message":"Uncaught PHP Exception TypeError: \"App\\Entity\\LogSystem\\LogLevel::from():
Argument #1 ($value) must be of type int, string given\" at /home/virtadpt/merivis.virtadpt.net/
    Part-DB-server/vendor/doctrine/orm/lib/Doctrine/ORM/Internal/Hydration/AbstractHydrator.php
line 709","context":{"exception":{"class":"TypeError","message":
"App\\Entity\\LogSystem\\LogLevel::from(): Argument #1 ($value) must be of type int,
string given","code":0,"file":"/home/virtadpt/merivis.virtadpt.net/Part-DB-server/vendor/
    doctrine/orm/lib/Doctrine/ORM/Internal/Hydration/AbstractHydrator.php:709"}},
"level":500,"level_name":"CRITICAL","channel":"request",
"datetime":"2023-11-18T21:40:00.575556-05:00","extra":{"token":{"authenticated":true,
    "roles":["ROLE_USER"],"user_identifier":"drwho"},"url":"/en/","ip":"my.home.ip.address",
    "http_method":"POST","server":"merivis.virtadpt.net",
    "referrer":"https://merivis.virtadpt.net/en/"}}
```

I strongly suspect that this is a configuration error, though I do not know where.  Perhaps there is a PHP module that is not enabled (my working hypothesis).  I know that the MySQL database configuration is correct because I can see my categories and the contents of the categories.  I can use the `bin/console` utility and not get any errors or failures (I've cleaned the caches multiple times, on the off chance that something was stuck in the cache and messing with me; no dice).

Output of `bin/console doctrine:migrations:current`:

```
DoctrineMigrations\Version20231114223101 - Add schema for part associations and vendor barcodes
```

Output of `bin/console partdb:check-requirements` (edited for space):
* [WARNING] You are using PHP 8.1.24. This will work, but a newer version is recommended.
* [OK] Opcache is enabled.
* [WARNING] Opcache configuration can be improved. See https://symfony.com/doc/current/performance.html for more info.
* [OK] PHP extension pdo_mysql is installed.
* [OK] PHP extension pdo_sqlite is installed.
* [OK] PHP extension curl is installed.
* [OK] PHP extension GD is installed.
* [OK] GD has jpeg support.
* [OK] GD has png support.
* [OK] GD has WebP support.
* [OK] Debug mode disabled.

```
[Part-DB-server]$ bin/console partdb:logs:show

 ! [NOTE] There are a total of 1698 log entries in the DB.                      

22:28:37 CRITICAL  [console] Error thrown while running command "partdb:logs:show".
Message: "App\Entity\LogSystem\LogLevel::from(): Argument #1 ($value) must be of type int,
string given" ["exception" => TypeError { …},"command" => "partdb:logs:show",
"message" => "App\Entity\LogSystem\LogLevel::from(): Argument #1 ($value) must be of type int,
string given"] ["token" => null,"command" => ["name" => "partdb:logs:show","arguments" =>
["command" => "partdb:logs:show"]]]

In ReflectionEnumProperty.php line 97:
                                                                               
  App\Entity\LogSystem\LogLevel::from(): Argument #1 ($value) must be of type int, string given 
```

jbtronics, 2023-11-19:
This is a pretty weird bug, which I can not explain why it happens at your system.
I can not replicate it, but I have added a commit to the master branch, which I hope should fix that problem.

virtadpt, 2023-11-19:
I'll give it a try when I get home tonight.  Thanks!

virtadpt, 2023-11-20:
Installed in hosting.  Logging in is no longer throwing 500's.  Pulling up entries is no longer throwing 500's and works as expected.  Clicking around the Tools menu seems to be working as expected.  Errors aren't showing up in var/log/prod.log.  I can edit existing entries without trouble.  Creating and deleting a new entry seems to work as expected, also.

I think it works.  Thank you so much.

closed 2023-11-20 by jbtronics