Part-DB/Part-DB-server #435: Can't revoke permissions for usergroup

cooki371, 2023-11-21:
**Describe the bug**
I can not revoke the delete permissions on the default user group 'user'.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'System - Groups'
2. Select 'usesr'
3. Navigate to 'Permissions'
4. Click all 'Delete' checkboxes until it is empty.
5. Save

**Expected behavior**
Changes should be saved. No 'Delete' right should reappear.

**Server Side**
- Part-DB Version: 1.8.2
- PHP Version: 8.1.23 (SAPI: fpm-fcgi)
- Database Server SQLite 3.34.1

**Desktop:**
 - OS: Windows 11
 - Browser Edge
 - Version 119.0.2151.72

**Additional context**
I want to prevent any dataloss thru to user errors. Therefore I want the users to not be able to delete parts and stuff, because the need of doing so would be a huge edge case in our process structure.

jbtronics, 2023-11-24:
This is the desired behavior, even if it might be a bit counterintuitive first.
The operations of the different permissions, are not independently but coupled to a certain degree. That means that you can not forbid operations, as long as you dont forbid operations which depends on it.

In your case, this means, that you can not forbid users to delete elements, as long as they are allowed to revert elements. Otherwise, they were able to revert an element back to its state before creation, which effectively means this element gets deleted.

To notify the admin of this behavior, the delete permission is set again. To forbid delete access, you need to uncheck the delete and "revert element" checkbox together, then the changes will be saved properly.

jbtronics, 2023-11-25:
I added an notice above the permission editor about this behavior and a flash warning is now shown, if the permissions get "corrected" that way. Hopefully that explains this behavior a bit better.

closed 2023-11-29 by jbtronics