Part-DB/Part-DB-server #436: Help Wanted: Cannot get SSO working with Keycloak behind a reverse proxy

Fuzzer11, 2023-11-23:
Hello!

**Describe the problem**
I have followed the guide to enable SSO and reviewed line by line, but I could'nt get the feature working. So I would like to ask for help.

**To Reproduce**
Steps to reproduce the environment:
1. The entire environment is running as a docker compose service stack.
2. I deployed a [preconfigured NGINX reverse proxy from LSIO](https://github.com/linuxserver/docker-swag).
3. I got a Keycloak instance working as described on the documentation connected to service running a postgreSQL databse.
4. I deployed a LDAP server and configured Keycloak user federation from it.
5. Finally I deployed an instance of Part-DB along with its database.
6. After I confirmed that every service are healthy I started following the [Part-DB guide to enable SSO](https://docs.part-db.de/installation/saml_sso.html).
7. First I configured a SAML client on Keycloak
8. Lastly I configured Part-DB

**Expected behavior**
Return the correct authentication after a succesful log in to the Identity Provider server. Instead I got a message: "An authentication exception occurred." on the web log in site after the redirection from the Keycloak log in page.

**Logs**
The message I consider the most relevant by reading the logs taken by docker is the following:

{"message":"The response was received at http://<SEVER_URL>/saml/acs instead of https://<SEVER_URL>/saml/acs","context":{},"level":400,"level_name":"ERROR","channel":"app","datetime":"2023-11-23T20:58:01.831071+00:00","extra":{"token":null,"url":"/saml/acs","ip":"172.20.0.2","http_method":"POST","server":"<SEVER_URL>","referrer":null}}

**Server Side**
- Part-DB Version: [e.g. Part-DB Version: 1.8.1

It might not be a difficult problem to solve, but I am not specialised in Identity Provider Services and though I tried some little changes in Keycloak configurations, nothing changed. I don't know what to do now.

Thanks.

jbtronics, 2023-11-24:
Have you configured the `TRUSTED_PROXIES` and the `DEFAULT_URI` env? These should be especially important in that case, so that the right request URLs can be determined.

Does the reverse proxy puts part db inside a folder/proxy? Then you would need to configure a prefix header too.

Also you should check that your reverse proxy sends a `x-forwarded-proto` header.

I think it should be possible to configure in keycloak what URLs are acceptable. However the better methods would to fix what kind of request URL Part-DB determines.

Fuzzer11, 2023-11-24:
Hello @jbtronics.Thank you for your answer. From your comments I realised that I missed to configure the TRUSTED_PROXIES variable, but that did not change anything. I have DEFAULT_URI configured as https://partdb.<SEVER_URL>.

So the URL to access it is through a subdomain instead of a subfolder.

The `x-forwarded-proto` header is enabled by default in the preconfigured proxy as well as some additional headers.

I forgot to mention that the logs I am showing are from Part-DB, I don't know what do you mean by "configure in Keycloak what URLs are acceptable".

Also, I could have not been looking at the right log entry, because I don't know if the cause of the authentication failure is the response in `http` instead of `https`.

The next entry in the log that includes the exception trace after that message is the following:

> {"message":"Authenticator failed.","context":{"exception":{"class":"Symfony\\Component\\Security\\Core\\Exception\\AuthenticationException","message":"The response was received at http://partdb.<SERVER_URL>/saml/acs instead of https://partdb.<SERVER_URL>/saml/acs","code":0,"file":"/var/www/html/vendor/nbgrp/onelogin-saml-bundle/src/Security/Http/Authenticator/SamlAuthenticator.php:95","trace":["/var/www/html/vendor/symfony/security-http/Authentication/AuthenticatorManager.php:176","/var/www/html/vendor/symfony/security-http/Authentication/AuthenticatorManager.php:158","/var/www/html/vendor/symfony/security-http/Authentication/AuthenticatorManager.php:140","/var/www/html/vendor/symfony/security-http/Firewall/AuthenticatorManagerListener.php:40","/var/www/html/vendor/symfony/security-http/Firewall/AbstractListener.php:26","/var/www/html/vendor/symfony/security-bundle/Security/LazyFirewallContext.php:60","/var/www/html/vendor/symfony/security-http/Firewall.php:128","/var/www/html/vendor/symfony/security-http/Firewall.php:95","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:260","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:220","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:56","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:158","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76","/var/www/html/vendor/symfony/http-kernel/Kernel.php:197","/var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/public/index.php:5"]},"authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":200,"level_name":"INFO","channel":"security","datetime":"2023-11-23T20:58:01.832007+00:00","extra":{"token":null,"url":"/saml/acs","ip":"172.20.0.2","http_method":"POST","server":"partdb.<SERVER_URL>","referrer":null}}

Thank you.

jbtronics, 2023-11-25:
With the latest commit (which will be part of the upcoming Part-DB 1.9) release, there is a new env option `SAML_BEHIND_PROXY`, which must be set if Part-DB is behind a reverse proxy and you wanna use SAML.
The SAML library has its own URL determination logic and require this setting to respect the reverse proxy headers.

Fuzzer11, 2023-12-07:
Hi @jbtronics thank you for your assistance and the last contributions. However I still get the same error when trying to log in through SSO despite I have recreated the containers after pulling the latest docker image and setting the `SAML_BEHIND_PROXY` variable to **true**. I am still missing something. Did you reproduce the envirnoment and get it working?

jbtronics, 2023-12-10:
This seemed to be caused by a wrong baseurl setting in the `nbgrp_onelogin_saml.yaml` config file. By default, it tried to autodetermine the right baseUrl out of the request object. But this seems to not properly work.

For now I changed it the way, that the `DEFAULT_URI` env is used too for this value, instead of the broken auto mechanism.

So if you use the latest `master` (or `edge`) tagged docker image and properly configure the `DEFAULT_URI` env, it should work now.

closed 2024-01-08 by jbtronics