Part-DB/Part-DB-server #437: Help needed installing Part-DB on MacOS in Docker

mrWheel, 2023-11-24:
**Describe the bug**
I followed the instructions for using the docker image but it seems the "Volumes" are not correct connected.

**To Reproduce**
Steps to reproduce the behavior:
1. Create the `docker-compose.yaml` as in [this document](https://docs.part-db.de/installation/installation_docker.html) example
2. In Docker Desktop go to `Resources` -> `File Sharing`
3. Mount the `db`, `public_media` and `uploads` folders, located in the same directory where the `docker-compose.yaml` file is located.
4. execute `docker-compose up --build`
     output:
```
     docker-compose up --build 
     [+] Building 0.0s (0/0)                                                                                         docker:desktop-linux
     [+] Running 1/0
     ✔ Container partdb  Created                                                                                                    0.0s 
     Attaching to partdb
     partdb  | [24-Nov-2023 08:04:24] NOTICE: fpm is running, pid 23
     partdb  | [24-Nov-2023 08:04:24] NOTICE: ready to handle connections
     partdb  | [24-Nov-2023 08:04:24] NOTICE: systemd monitor interval set to 10000ms
     partdb  | AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.31.0.2. Set the 
     'ServerName' directive globally to suppress this message
     partdb  | AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.31.0.2. Set the 
     'ServerName' directive globally to suppress this message
     partdb  | [Fri Nov 24 08:04:24.347874 2023] [mpm_event:notice] [pid 1:tid 139697104616768] AH00489: Apache/2.4.56 (Debian) configured -- resuming normal operations
     partdb  | [Fri Nov 24 08:04:24.348264 2023] [core:notice] [pid 1:tid 139697104616768] AH00094: Command line: 'apache2 -D FOREGROUND'
```
6. execute `docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate`
7. in browser `http://localhost:8080`

**Expected behavior**
Part-DB page starts -> OK
Expect `app.db` in `./db` --> **it is not**!!!
There is no `app.db` anywhere on my iMac!

**Screenshots**
[image]

[image]

**Server Side**
- Part-DB Version: [Part-DB 1.8.2]
- Database Server [SQLite]

**Desktop:**
 - OS: [macOS Monterey 12.7.1]
 - Browser [chrome]
 - Version [119.0.6045.159 (Official Build) (x86_64)]

**Additional context**
I have had a working Part-DB (I created it long ago with the provided `Dockerfile`, but that does not work anymore complaining about `symfony.conf` not found) in Docker but suddenly Docker would not startup anymore. So I followed [these](https://stackoverflow.com/questions/57877600/docker-compose-creates-directory-instead-of-file) instructions.

> I had the same issue on my Mac. It was working well, suddenly, one day it stops working. I was able to run it again by the following method:
> 
> 1. Go to finder and open Applications > Utilities > Activity Monitor.app
> 2. Find Docker or Docker Desktop (it was in Not responding mode in my case) and Force Kill the app.
> 3. Run the following command in Terminal:
>           sudo rm -rf ~/Library/Containers/com.docker.*
> 
> 4. Open the setting file in the terminal:
> 
>          sudo nano ~/Library/Group\ Containers/group.com.docker/settings.json
> 
> then find "filesharingDirectories":
> 
> 5. Remove all entries inside this key and leave it as "filesharingDirectories": [], save and close the file.
> 6. Finally go to the Application folder, find Docker and run it (wait sometime, it takes a minute or so to run)

Probably steps 3, 4, and 5 removed my working Part-DB container!!!
Luckily I still have the `app.db` so I hope to restore my data.

jbtronics, 2023-11-25:
> * In Docker Desktop go to `Resources` -> `File Sharing`
> * Mount the `db`, `public_media` and `uploads` folders, located in the same directory where the `docker-compose.yaml` file is located.

You should not need to do this steps when using a docker-compose file. These mapping should be done automatically, when running `docker-compose up`.

I would expect, that defining these manually before running docker-compose, messes up the real mapping process.
So you should try to just run `docker-compose up` in a fresh folder with the `docker-compose.yaml` file.
The folders and volume mappings should get created automatically.

mrWheel, 2023-11-25:
@jbtronics, Jan,
Thanks for your reply!
I removed the bindings from `Resources` -> `File Sharing`
If I run `docker-composer up -d` I get this messages:
```
% docker-compose up -d
  [+] Building 0.0s (0/0)                                                                                                docker:desktop-linux
  [+] Running 0/0
   ⠋ Container partdb  Starting                                                                                                          0.0s 
  Error response from daemon: Mounts denied: 
  The path /Users/x/myDockerfiles/Part-DB-symfony-master2/db is not shared from the host and is not known to 
  Docker.
  You can configure shared paths from Docker -> Preferences... -> Resources -> File Sharing.
  See https://docs.docker.com/desktop/mac for more info.
```
This is my docker-compose.yaml file:
```
version: '3.3'
services:
  partdb:
    #user: "root"
    container_name: partdb
    # By default Part-DB will be running under Port 8080, you can change it here
    ports:
      - '8080:80'
    volumes:
      # By default
      - ./db:/var/www/html/var/db
      - ./uploads:/var/www/html/uploads
      - ./public_media:/var/www/html/uploads/media
    restart: unless-stopped
    image: jbtronics/part-db1:latest
    environment:
      # Put SQLite database in our mapped folder. You can configure some other kind of database here too.
      - DATABASE_URL=sqlite:///%kernel.project_dir%/var/db/app.db
      # In docker env logs will be redirected to stderr
      - APP_ENV=docker

      # You can configure Part-DB using environment variables
      # Below you can find the most essential ones predefined
      # However you can add add any other environment configuration you want here
      # See .env file for all available options or https://docs.part-db.de/configuration.html

      # The language to use serverwide as default (en, de, ru, etc.)
      - DEFAULT_LANG=en
      # The default timezone to use serverwide (e.g. Europe/Berlin)
      - DEFAULT_TIMEZONE=Europe/Berlin
      # The currency that is used inside the DB (and is assumed when no currency is set). This can not be changed later, so be sure to set it the currency used in your country
      - BASE_CURRENCY=EUR
      # The name of this installation. This will be shown as title in the browser and in the header of the website
      - INSTANCE_NAME=Part-DB

      # Allow users to download attachments to the server by providing an URL
      # This could be a potential security issue, as the user can retrieve any file the server has access to (via internet)
      - ALLOW_ATTACHMENT_DOWNLOADS=0
      # Use gravatars for user avatars, when user has no own avatar defined
      - USE_GRAVATAR=0

      # Override value if you want to show to show a given text on homepage.
      # When this is empty the content of config/banner.md is used as banner
      - BANNER=Parts Database system van Willem Aandewiel<br>Build with docker-compose.yaml
```