Part-DB/Part-DB-server #443: 500 Error after Upgrade from 1.8.2 to 1.9.0

Pyromane, 2023-11-26:
I have PartDB running in a Docker with MySQL.
I have updated the container and then
php bin/console doctrine:migrations:migrate
as well as a
php bin/console cache: clear
executed

However, I get a 500 error message with different browsers.
The following error message caught my eye in the log:
`{"message":"Failed to save key \"version\" of type Version\\Version: fopen(/var/www/html/var/cache/docker/pools/system/pbl-hE1W7B/sJcgOlrz): Failed to open stream: Permission denied","context":{"key":"version","exception":{"class":"ErrorException","message":"fopen(/var/www/html/var/cache/docker/pools/system/pbl-hE1W7B/sJcgOlrz): Failed to open stream: Permission denied","code":0,"file":"/var/www/html/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:95","trace":["/var/www/html/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:95","/var/www/html/vendor/symfony/cache/Adapter/PhpFilesAdapter.php:248","/var/www/html/vendor/symfony/cache/Adapter/AbstractAdapter.php:148","/var/www/html/vendor/symfony/cache/Traits/AbstractAdapterTrait.php:239","/var/www/html/vendor/shivas/versioning-bundle/src/Service/VersionManager.php:116","/var/www/html/vendor/shivas/versioning-bundle/src/Twig/VersionExtension.php:31","/var/www/html/vendor/twig/twig/src/ExtensionSet.php:331","/var/www/html/vendor/twig/twig/src/Environmen
t.php:787","/var/www/html/vendor/twig/twig/src/Environment.php:800","/var/www/html/vendor/twig/twig/src/Template.php:367","/var/www/html/vendor/twig/twig/src/Template.php:379","/var/www/html/vendor/twig/twig/src/TemplateWrapper.php:38","/var/www/html/vendor/twig/twig/src/Environment.php:280","/var/www/html/vendor/symfony/twig-bridge/ErrorRenderer/TwigErrorRenderer.php:54","/var/www/html/vendor/symfony/error-handler/ErrorRenderer/SerializerErrorRenderer.php:65","/var/www/html/vendor/symfony/http-kernel/Controller/ErrorController.php:41","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:181","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:76","/var/www/html/vendor/symfony/http-kernel/EventListener/ErrorListener.php:108","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:260","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:220","/var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php:56","/var/www/html/vendor/symfony/http-kernel/HttpKernel.php:239","/
var/www/html/vendor/symfony/http-kernel/HttpKernel.php:91","/var/www/html/vendor/symfony/http-kernel/Kernel.php:197","/var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/public/index.php:5"]},"cache-adapter":"Symfony\\Component\\Cache\\Adapter\\PhpFilesAdapter"},"level":300,"level_name":"WARNING","channel":"cache","datetime":"2023-11-26T19:49:12.893671+00:00","extra":{"token":null,"url":"/","ip":"192.168.0.30","http_method":"GET","server":"192.168.0.106","referrer":null}}`

Full log is attached
[_partdb_logs.txt](https://github.com/Part-DB/Part-DB-server/files/13467956/_partdb_logs.txt)

jbtronics, 2023-11-26:
Thats an cache permission issue, that happens if you have not run the cache clear command as the web user (but as root or something else). Therefore the webserver running as another user can not write the cache files anymore.

This can be fixed by either running a chown or by simply deleting the `var/cache` folder in Part-DB

Pyromane, 2023-11-26:
Thanks for the quick feedback!

I had now completely restored 1.8.2 again as a test and proceeded from there as described above, except that this time it worked without any problems and I no longer received an error message

closed 2023-11-26 by Pyromane

influx3k, 2023-12-08:
Hi, I'm getting I think this same issue.  I tried upgrading to 1.10.0; didn't work, so I tried reverting back to 1.9.1; now that doesn't work either.  I think I'm getting the same error:

```
10.0.0.117 - - [08/Dec/2023:16:24:01 +0000] "GET / HTTP/1.1" 500 4282 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36"
10.0.0.117 - - [08/Dec/2023:16:24:53 +0000] "-" 408 0 "-" "-"
200 Matched route "redirector". | context={"route":"redirector","route_parameters":{"_route":"redirector","_controller":"App\\Controller\\RedirectController::addLocalePart","url":""},"request_uri":"http://10.0.0.126:8080/","method":"GET"} level_name=INFO channel=request datetime=2023-12-08T16:25:25.980464+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Checking for authenticator support. | context={"firewall_name":"main","authenticators":5} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985024+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Checking support on authenticator. | context={"firewall_name":"main","authenticator":"App\\Security\\ApiTokenAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985038+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Authenticator does not support the request. | context={"firewall_name":"main","authenticator":"App\\Security\\ApiTokenAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985048+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Checking support on authenticator. | context={"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985055+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Authenticator does not support the request. | context={"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985064+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Checking support on authenticator. | context={"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985070+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Authenticator does not support the request. | context={"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985076+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Checking support on authenticator. | context={"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985081+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Authenticator does not support the request. | context={"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985088+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Checking support on authenticator. | context={"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985093+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Remember-me cookie detected. | context={} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985122+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
100 Remember-me cookie detected. | context={} level_name=DEBUG channel=security datetime=2023-12-08T16:25:25.985853+00:00 extra={"token":null,"url":"/","ip":"10.0.0.117","http_method":"GET","server":"10.0.0.126","referrer":null}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:211","trace":["/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:102","/var/www/html/vendor/doctrine/d [... cut ...] er":null}}
- -  08/Dec/2023:16:25:25 +0000 "GET /index.php" 500
10.0.0.117 - - [08/Dec/2023:16:25:25 +0000] "GET / HTTP/1.1" 500 4282 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36"
```

I tried deleting the `var/cache` folder in the container and running both `php bin/console cache:clear` and `php bin/console doctrine:migrations:migrate`.  Still get a 500 error.  Tried restarting the container and can't get it to work at all.

jbtronics, 2023-12-08:
> I tried deleting the `var/cache` folder in the container and running both `php bin/console cache:clear` and `php bin/console doctrine:migrations:migrate`. Still get a 500 error. Tried restarting the container and can't get it to work at all.

You must execute these commands as the webserver user. Otherwise the permissions of the cache files get wrong and you end up with this error.

jbtronics, 2023-12-08:
`docker exec --user=www-data partdb php bin/console cache:clear`

influx3k, 2023-12-08:
Got it.  I deleted the file and now it works.  Weird because I did not do that; I did not change any permissions or anything like that.

influx3k, 2023-12-08:
Thanks for your help!!!