Part-DB/Part-DB-server #447: average cost of parts in stock

bdale, 2023-12-01:
As a small business owner with an electronics part inventory, a feature of partkeepr I got used to having was average price for each part I have in inventory.   I don't see any equivalent in Part-DB, and I note that this information seems to have been dropped when importing my existing partkeepr database.

The price data was collected by including the price (per item or per lot)  when stock was added, and for each part in the datbase the average price of all items currently in stock was maintained. 

While I understand that many times what matters is what it would cost to get more of a part, for tax accounting purposes and understanding the actual cost of goods sold that are built from parts in existing inventory, knowing what the parts on hand cost at the time they were purchased is important.

Am I just missing something about how this information is stored in Part-DB today?  If not, I'd love to see this ability added.

jbtronics, 2023-12-01:
Part-DB does not store the average price in database, but calculate it dynamically out of the price infos:
[image]

The yellow marked values, are the average price range. So basically the average price for the maximum acquireable amount (so with the highest discount) to the average price for the miniumm acquireable amount (e.g. 1 with the lowest discount). The prices are averaged across all suppliers/distributors.

henrikkoch, 2024-01-05:
Can you somehow see the value of your whole inventory (I mean the total amount/value)?

LtMost, 2024-01-25:
Would it be possible to get the average price of a part in the BOM or an average price of the Item if i would use the parts in the BOM "X" times.

CyberTeckyLLC, 2024-11-23:
Greetings,

I would like to add to these good questions regarding Part-DB.  Actually, I posted a similar question in the 'Discussions' section of the Part-DB GitHub, so if this may cause some redundancy, I apologize.  I'm just hoping to figure out a way to manage my own inventory a bit more efficiently, specifically with tracking inputs of part cost estimates of my own inventory.  Please read the following 'Discussions' post [here](https://github.com/Part-DB/Part-DB-server/discussions/777#discussion-7530232).

Thank you!

Chad