Part-DB/Part-DB-server #456: Part-db complain about 404 Not Found but is looking in the wrong address

amitzioni, 2023-12-12:
After building the new part-db and migrating the database from legacy (0.5.15) to current version (1.10.0), I still can't get part-db to work. 
The homepage is on **https://crm/lager110/public** shows great and I'm also able to login. However when I click on one of the categories on the left menu I get an "Not Found (404 Error)" pop up. However the url that the message refers to is completely weird. An example of such weird url is **https://crm/lager110/public/lager110/public/en/category/143/parts**. 
It seems like that the /lager110/public/ just repeats itself. 

I'm using apache2 and mariadb

[image]

jbtronics, 2023-12-12:
You should configure your webserver the way, that the public folder is the DocumentRoot of your webhost. This is required to ensure that users can not access folders containing sensitive files. Also it makes it really difficult to determine the right URL pathes...
See https://docs.part-db.de/installation/installation_guide-debian.html#configure-apache2-to-show-part-db for that.

The easiest way is, if you put on its own vhost. In principle it should also be possible to put Part-DB in a subpath of a webserver (like via a reverse proxy). But that is not so well tested and even then you should avoid to expose Part-DB internal structure with the public folder.

amitzioni, 2023-12-12:
At the end I put it in a separate virtual host, as written in the manual. 
Thank you !

closed 2023-12-12 by amitzioni