Part-DB/Part-DB-server #457: Javascript error when saving a part

NomAnor, 2023-12-15:
**Describe the bug**
I get a javascript error in the browser console when I
* Save part after editing
* Open new part page
* Open user page
* Open group page
* Open any page under edit or tools

```
TypeError: e is undefined
    create http://localhost:9122/build/9132.f2a1f55f.js:2
    promise callback*create http://localhost:9122/build/9132.f2a1f55f.js:2
    value http://localhost:9122/build/4942.d6927c4d.js:2
    connect http://localhost:9122/build/8139.3177793d.js:2
    connectContextForScope http://localhost:9122/build/8139.3177793d.js:2
    scopeConnected http://localhost:9122/build/8139.3177793d.js:2
    elementMatchedValue http://localhost:9122/build/8139.3177793d.js:2
    tokenMatched http://localhost:9122/build/8139.3177793d.js:2
    tokenMatched http://localhost:9122/build/8139.3177793d.js:2
    tokensMatched http://localhost:9122/build/8139.3177793d.js:2
    tokensMatched http://localhost:9122/build/8139.3177793d.js:2
    elementMatchedAttribute http://localhost:9122/build/8139.3177793d.js:2
    elementMatched http://localhost:9122/build/8139.3177793d.js:2
    addElement http://localhost:9122/build/8139.3177793d.js:2
    processTree http://localhost:9122/build/8139.3177793d.js:2
    processAddedNodes http://localhost:9122/build/8139.3177793d.js:2
    processMutation http://localhost:9122/build/8139.3177793d.js:2
    processMutations http://localhost:9122/build/8139.3177793d.js:2
    mutationObserver http://localhost:9122/build/8139.3177793d.js:2
    x http://localhost:9122/build/8139.3177793d.js:2
    x http://localhost:9122/build/8139.3177793d.js:2
    render http://localhost:9122/build/8139.3177793d.js:2
    renderSnapshot http://localhost:9122/build/8139.3177793d.js:2
    render http://localhost:9122/build/8139.3177793d.js:2
    loadFrameResponse http://localhost:9122/build/8139.3177793d.js:16
    loadResponse http://localhost:9122/build/8139.3177793d.js:16
    requestSucceededWithResponse http://localhost:9122/build/8139.3177793d.js:16
    receive http://localhost:9122/build/8139.3177793d.js:2
    perform http://localhost:9122/build/8139.3177793d.js:2
    visit http://localhost:9122/build/8139.3177793d.js:16
    visit http://localhost:9122/build/8139.3177793d.js:16
    loadSourceURL http://localhost:9122/build/8139.3177793d.js:16
    sourceURLChanged http://localhost:9122/build/8139.3177793d.js:16
    attributeChangedCallback http://localhost:9122/build/8139.3177793d.js:2
    set src http://localhost:9122/build/8139.3177793d.js:2
    navigateFrame http://localhost:9122/build/8139.3177793d.js:16
    withCurrentNavigationElement http://localhost:9122/build/8139.3177793d.js:16
    navigateFrame http://localhost:9122/build/8139.3177793d.js:16
    linkClickIntercepted http://localhost:9122/build/8139.3177793d.js:16
    linkClicked http://localhost:9122/build/8139.3177793d.js:2
    C http://localhost:9122/build/8139.3177793d.js:2
    notifyApplicationAfterClickingLinkToLocation http://localhost:9122/build/8139.3177793d.js:16
    applicationAllowsFollowingLinkToLocation http://localhost:9122/build/8139.3177793d.js:16
    willFollowLinkToLocation http://localhost:9122/build/8139.3177793d.js:16
    clickBubbled http://localhost:9122/build/8139.3177793d.js:2
    clickCaptured http://localhost:9122/build/8139.3177793d.js:2
4942.d6927c4d.js:2:70489
    value http://localhost:9122/build/4942.d6927c4d.js:2
    (Async: promise callback)
    value http://localhost:9122/build/4942.d6927c4d.js:2
    connect http://localhost:9122/build/8139.3177793d.js:2
    connectContextForScope http://localhost:9122/build/8139.3177793d.js:2
    scopeConnected http://localhost:9122/build/8139.3177793d.js:2
    elementMatchedValue http://localhost:9122/build/8139.3177793d.js:2
    tokenMatched http://localhost:9122/build/8139.3177793d.js:2
    tokenMatched http://localhost:9122/build/8139.3177793d.js:2
    tokensMatched http://localhost:9122/build/8139.3177793d.js:2
    tokensMatched http://localhost:9122/build/8139.3177793d.js:2
    elementMatchedAttribute http://localhost:9122/build/8139.3177793d.js:2
    elementMatched http://localhost:9122/build/8139.3177793d.js:2
    addElement http://localhost:9122/build/8139.3177793d.js:2
    processTree http://localhost:9122/build/8139.3177793d.js:2
    processAddedNodes http://localhost:9122/build/8139.3177793d.js:2
    processMutation http://localhost:9122/build/8139.3177793d.js:2
    processMutations http://localhost:9122/build/8139.3177793d.js:2
    mutationObserver http://localhost:9122/build/8139.3177793d.js:2
    (Async: FrameRequestCallback)
    x http://localhost:9122/build/8139.3177793d.js:2
    x http://localhost:9122/build/8139.3177793d.js:2
    render http://localhost:9122/build/8139.3177793d.js:2
    renderSnapshot http://localhost:9122/build/8139.3177793d.js:2
    render http://localhost:9122/build/8139.3177793d.js:2
    loadFrameResponse http://localhost:9122/build/8139.3177793d.js:16
    loadResponse http://localhost:9122/build/8139.3177793d.js:16
    requestSucceededWithResponse http://localhost:9122/build/8139.3177793d.js:16
    receive http://localhost:9122/build/8139.3177793d.js:2
    perform http://localhost:9122/build/8139.3177793d.js:2
    visit http://localhost:9122/build/8139.3177793d.js:16
    visit http://localhost:9122/build/8139.3177793d.js:16
    loadSourceURL http://localhost:9122/build/8139.3177793d.js:16
    sourceURLChanged http://localhost:9122/build/8139.3177793d.js:16
    attributeChangedCallback http://localhost:9122/build/8139.3177793d.js:2
    set src http://localhost:9122/build/8139.3177793d.js:2
    navigateFrame http://localhost:9122/build/8139.3177793d.js:16
    withCurrentNavigationElement http://localhost:9122/build/8139.3177793d.js:16
    navigateFrame http://localhost:9122/build/8139.3177793d.js:16
    linkClickIntercepted http://localhost:9122/build/8139.3177793d.js:16
    linkClicked http://localhost:9122/build/8139.3177793d.js:2
    C http://localhost:9122/build/8139.3177793d.js:2
    notifyApplicationAfterClickingLinkToLocation http://localhost:9122/build/8139.3177793d.js:16
    applicationAllowsFollowingLinkToLocation http://localhost:9122/build/8139.3177793d.js:16
    willFollowLinkToLocation http://localhost:9122/build/8139.3177793d.js:16
    clickBubbled http://localhost:9122/build/8139.3177793d.js:2
    (Async: EventListener.handleEvent)
    clickCaptured http://localhost:9122/build/8139.3177793d.js:2
```

**Server Side**
- Part-DB Version: 1.10.1 (docker)
- PHP Version:  	8.1.26 (SAPI: fpm-fcgi)
- Database Server: sqlite (3.34.1)

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox
 - Version: 120.0

jbtronics, 2023-12-15:
This error can basically be ignored. 
It is thrown somewhere inside CKEditor (the rich text editor), most likely because an already existing CKEditor field gets initilaized again. Thats why this error occurs on every page, which has a rich text editor field.

However this issues does not affect anything as far as I am aware.

jbtronics, 2024-01-15:
This is fixed with Part-DB 1.10.3

closed 2024-01-15 by jbtronics