Part-DB/Part-DB-server #458: Unintuitive behaviour of sidebar items

NomAnor, 2023-12-16:
**Describe the bug**
When you click on a sidebar item, it gets selected (blue background) and the corresponding page is loaded.
When you click again on the selected item, it gets deselected and nothing happens. I find this a weird behaviour for a navigation sidebar. I would expect that every click causes the correspondig page to load and the already selected page to just reload.

Also the selection seems to be per group. I can have a blue background on a category item and a tool item whereas only the current page should be highligted in the navigation.

**Background**
When I look at a part info page and I want to go back to the category, I often click on the category again, but nothing happens because it gets deselected.

**Server Side**
- Part-DB Version: 1.10.1 (docker)
- PHP Version:  	8.1.26 (SAPI: fpm-fcgi)
- Database Server: sqlite (3.34.1)

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox
 - Version: 120.0

Tolomaj, 2024-01-05:
I got same bug happening

**Server Side**
- Part-DB Version: 1.10.1
- PHP Version: PHP 8.1.27

**Desktop (please complete the following information):**
- OS: Windows 12
- Browser: Google Chrome
- Version: 120.0.6099.131

jbtronics, 2024-01-27:
This should be fixed now with the latest changes. The page gets directly called, if you click on an already selected node.

Additionally, a node now should get automatically unselected, if you navigate away from the page which the node refers.

closed 2024-01-27 by jbtronics