Part-DB/Part-DB-server #459: DigiKey API

workino, 2023-12-17:
Digikey provider is having a bug during the search. Instead to search the keyword given is searching for something completely different. Ex. i searched for tda7052at and the result was a electrolytic capacitor.
Also the api url now looks to be the sandbox only. 

**To Reproduce**
Steps to reproduce the behavior:
1. modify all the file to use sandbox instead of api...
2. make a search
3. result is not what expected

**Expected behavior**
the search will give the expected result 

**Screenshots**
Actually i cannot because i must change again all the url.

**Server Side**
- Part-DB Version: 1.10.1
- PHP Version: docker version
- Database Server sqlite

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser chrome / firefox

**Additional context**
also octopart is not working as expected.
Thanks

jbtronics, 2023-12-18:
> Steps to reproduce the behavior:
> 
>     1. modify all the file to use sandbox instead of api...
> 
>     2. make a search
> 
>     3. result is not what expected

If you explicitly modify Part-DB to use digikeys sandbox API endpoint (which you should not do), then you the result you have observed is correct.
Digikeys sandbox returns this capacitor for all querys, no matter what keyword. Thats not an bug by Part-DB, but an design choice of Digikey. In 99,99% you want to use the real Digikey endpoint, instead of the sandbox.

The results of the providers are cached, to use less API requests. So if you change anything on the providers code (or their settings), you need to clear Part-DB cache afterwards, or you may end up with old results, which not reflect your changes.

workino, 2023-12-18:
> If you explicitly modify Part-DB to use digikeys sandbox API endpoint (which you should not do), then you the result you have observed is correct. Digikeys sandbox returns this capacitor for all [...]

Great thank for your answer. Can you please help me to understand how to use the correct api? the only i found registering on digikey was the sandbox.

Thanks

jbtronics, 2023-12-18:
You dont need (and must not) modify any of Part-DB code.
 
Just go to the digikey developer portal, go to "organizations" in the right corner, and create an organisation.
Using this organisation you can create production apps.

workino, 2023-12-18:
> You dont need (and must not) modify any of Part-DB code.
> 
> Just go to the digikey developer portal, go to "organizations" in the right corner, and create an organisation. Using this organisation you can create production apps.
> 
> The ClientID and Secret you receive, must be put into the env options of Part-DB

Shame on me :) i did it just before your message. Probably can be interesting to highlight this procedure in the provider documentation.

Thanks for your help, all looks working correctly now

closed 2024-01-08 by jbtronics