Part-DB/Part-DB-server #464: Adding a part via the API just by a manufacturer number?

lmoe, 2023-12-23:
Hey, thanks a lot for this great software!

**Is your feature request related to a problem? Please describe.**
Not really a usability problem at all.

**Describe the solution you'd like**
I buy my components from Digikey. Their packages contain a data matrix which contains the quantity and the manufacturer product number. I wrote a quick PoC website for phones that can scan the code, parse it and send that information to PartDB. 

Now I've noticed that there is no API to just "add or update" a part just by its manufacturer number which also includes fetching data from InfoProviders such as DigiKey. Furthermore, providers are not exposed via API either. So I'm stuck a bit. :D

Right now, my phone "app" would need additional API access keys to Digikey and would need to sort out everything by itself, then post the part data to PartDB.

You already offer something like this via `/en/part/from_info_provider/digikey/manufacturer-number-12345/create` but this is not exposed via the API.

I would propose a similar API route which is a bit more invasive. For example it would automatically create the categories pulled from the info provider. So external apps can just POST the manufacturer number to the API with a certain quantity - maybe including the category - and PartDB automatically fetches the data from an info provider and inserts/updates the data. I guess most of the logic is already there.

Or as an alternative, just return every info that `from_info_provider` would gather as JSON over the API, so the user can inspect it. The part can then be created or updated with the returned data it in the next step.

Is this something worth adding? I think it would make using it a bit more simple. Initially, when I installed PartDB, I had the impression that the current scanning functionality already can do this. 

You can find my PoC here: https://github.com/lmoe/partdb-digikey-scanner
For now a very simple svelte app, but can be trimmed down more.

jbtronics, 2024-01-15:
Exposing the info provider datastructures should be relatively easy to implement via the API.

Directly having API endpoint to create a part from it, is most likely more difficult. And i dont know how useful it is, as I guess you would want to review/edit the data in most cases.

CRSLEE, 2025-01-27:
I would like to scan e.g. LCSC labels, extract the Cxxxx number and add a new part via the API.
Are there any news regarding this topic?

greetings