Part-DB/Part-DB-server #465: Treatment of Greek small letters MU and MICRO characters as the same character in the search.

RaptorDE, 2023-12-25:
**Describe the bug**

When I create a capacitor with a capacitance of 100µF on Windows and then search for it on Android, I get no results.

The reason for this is that I created the micro-character (Unicode number U+00B5) under Windows with the key combination [Alt Gr] + [M]. On Android, however, the Greek minuscule "μ" (Unicode number U+03BC) is generated with the Google keyboard.

This behavior is theoretically correct, but the end user cannot distinguish U+00B5 from U+03BC.

**Expected behavior**
Treatment of Greek small letters MU and MICRO characters as the same character in the search.

There are probably many other characters that look the same or very similar with different Unicode.

Here is the Wikipedia entry for μ
https://en.m.wikipedia.org/wiki/Mu_(letter)

**Server Side**
- Part-DB Version: [1.10.1] Docker

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Browser [Firefox]

**Smartphone (please complete the following information):**
 - Device: [Samsung S10]
 - OS: [Android 12]
 - Browser [Chrome]

jbtronics, 2024-01-08:
I will think about how to properly fix this behavior. But that might be difficult, as the database treats them (correctly) as different characters.

RaptorDE, 2024-01-14:
I don't have much experience with web frontend development. Could a correction in input sanitization solve the problem?
In Python I can use the function [`unicodedata.normalize(form, unistr)`](https://docs.python.org/3/library/unicodedata.html#unicodedata.normalize) to convert a string to the [Unicode Normalization Forms](http://unicode.org/reports/tr15/). I don't know if it works that easily with php?

Here is a small Python script that converts two Unicoe characters into the different normalization forms and makes them comparable.

Output:
```
Normalisierungsform:      Zeichen A: µ Hex: 0xb5  Zeichen B: μ Hex: 0x3bc
Normalisierungsform: NFD  Zeichen A: µ Hex: 0xb5  Zeichen B: μ Hex: 0x3bc
Normalisierungsform: NFC  Zeichen A: µ Hex: 0xb5  Zeichen B: μ Hex: 0x3bc
Normalisierungsform: NFKD Zeichen A: μ Hex: 0x3bc Zeichen B: μ Hex: 0x3bc
Normalisierungsform: NFKC Zeichen A: μ Hex: 0x3bc Zeichen B: μ Hex: 0x3bc
```
```
import unicodedata

zeichenA = "µ"
zeichenB = "μ"

## http://unicode.org/reports/tr15/#Normalization_Forms_Table
## https://docs.python.org/3/library/unicodedata.html#module-unicodedata

# ohne Normalisierung
def ausgabe(Normalisierungsform, zA, zB):
    a = "Normalisierungsform: " + Normalisierungsform.ljust(5)+ "Zeichen A: "+ zeichenA +  " Hex: " + hex(ord(zeichenA)).ljust(5) + " Zeichen B: "+ zeichenB + " Hex: " + hex(ord(zeichenB))
    return a

# Normalisierungsform D (NFD) Kanonische Zersetzung
# Normalisierungsformular C (NFC) Kanonische Zersetzung gefolgt von Canonical Composition
# Normalisierungsform KD (NFKD) Kompatibilität
# Normalisierungsform KC (NFKC) Kompatibilität gefolgt von Canonical Composition
Normalisierung = ["NFD","NFC","NFKD","NFKC"]

def ausgabeNormalisierung(Normalisierungsform, zA, zB):
    a = "Normalisierungsform: " + Normalisierungsform.ljust(5)+ "Zeichen A: "+ unicodedata.normalize(Normalisierungsform,zA) +  " Hex: " + hex(ord(unicodedata.normalize(Normalisierungsform,zA))).ljust(5) + " Zeichen B: "+ unicodedata.normalize(Normalisierungsform,zB) + " Hex: " + hex(ord(unicodedata.normalize(Normalisierungsform,zB)))
    return a

print(ausgabe("",zeichenA,zeichenB))
for normalisierungstyp in Normalisierung:
    print(ausgabeNormalisierung(normalisierungstyp,zeichenA,zeichenB))
```

jbtronics, 2024-01-15:
What kind of database do you use?

I tested it on the demo instance (which uses MySQL) and it makes no difference there. (And the used coalition there should be quite wide).
It is possible that Sqlite has a different behavior.

The problem is that we need to do the searches in the database for performance reasons. Therefore it would not help much, if we perform normalizations in PHP for comparision, as the database values would be needed to be normalized the same way.

RaptorDE, 2024-01-16:
I didn't think about trying it in the demo instance. 

I noticed the problem with the SQLite database.
I have now migrated my data into a MySQL database and can confirm that it works there. Since I wanted to use a MySQL or mariaDB for my production system anyway, this would solve the problem for me. 

Maybe you can write this in the documentation as an advantage for the MySQL data base?

closed 2024-01-16 by RaptorDE

jbtronics, 2024-01-16:
I added such a hint to documentation