Part-DB/Part-DB-server #472: Population of categories

SANCLA, 2024-01-08:
First, let me tell you that I am a big fan of part-db.
Until a few years ago I still used partkeepr, but this application can now be buried with. Meanwhile, I experimented with some other apps, but it were mostly too complex, not easy to understand or broken installations.
Until I discovered this app. My thanks for all the time and effort put into this.

Since abandoning partkeepr, I have reverted to a more practical system with labels on boxes. That worked quite nicely until I moved 2 times, looting things in between. Moved parts with unknown destination to other boxes and crates. The labels also fell off in some places, that didn't help either.

So, time for a fresh, clean and probably time-consuming new start...
The upside is that I can make different choices, not necessarily better, but one can try...

What I am running into now is a choice regarding a good arrangement of categories. I have more then 1000 small electronic parts and without categories it is difficult to get a good overview. 

**Describe the solution you'd like**
Now, my request is, if it is possible to add things like categories and maybe vendors more easily with a the option to prepopulate at least the categories. For now, this is a feature request, but at the same time a help request if anyone has a good base that I can adopt. Somehow I suspect Octopart could be a good standard, but I lack the experience with it to come up with good arguments.

**Describe alternatives you've considered**
I searched the Internet for a good base, which I could get a good start with by using the import function of part-db. It may be due to my search keywords, but that search has not been very successful so far. 

Another option I have considered is to manually copy all categories from octoparts. 
That turns out to be pretty hellish work. 

Given there is an import functionality for this in part-db, I suspect this could be easier. 
I searched open and closed issues on this github, could not find any raised issue about this topic.
If anyone could point me in the right direction, I would be genuinely grateful.

**Additional context**
Irrelevant

jbtronics, 2024-01-09:
I already thought about providing some presets for the datastructures (categories, manufacturers, etc.) and this is planned for the future.
But i guess its not that easy, as everybody has its own requirements. I guess its not so problematic, if you have some manufacturers in your database, which you dont use.
However I use categories mostly to navigate through the database, and i guess it would not be good for UX if you have hundrets of category entries, which are empty, because you just dont have the parts.

I guess octopart has a pretty detailed category system, as it has to manage millions of parts. We have to find the balance between having a powerful detailed presets and being not overly complex for hobbyist use.

And just as a note. You can create new categories on the fly while creating new parts, by just typing the new category name in the category field. This also works with complex categories structures.
That might make it a lot easier to create the required category structure for you (if you know what you want)

ECCOsea, 2024-01-24:
May be you talking about parameters? Categories already available in this base, and works ok. I think your request more fit to parameter search as we have on mouser and etc electronic providers. For example: [Capacitors](https://eu.mouser.com/c/passive-components/capacitors/ceramic-capacitors/mlccs-multilayer-ceramic-capacitors/multilayer-ceramic-capacitors-mlcc-smd-smt/)

You can search your parts by parameters (manufacturer present also as parameter)

I have opened this theme already in discussion [here](https://github.com/Part-DB/Part-DB-server/discussions/303)

> However I use categories mostly to navigate through the database, and i guess it would not be good for UX if you have hundrets of category entries, which are empty, because you just dont have the parts.
For now categories works fine. And here is basic search now implemented! (may be add some checkbox for hide non relevant categories when performing search).