Part-DB/Part-DB-server #477: Fresh install with migradated database, calling index.php two times: Error calling /index.php/index.php/en/category/2/parts.

ObligedTester, 2024-01-14:
**Describe the bug**
A clear and concise description of what the bug is.

Fresh install with migrated database.
The first page loads as expected, all my categories are shown.
However, clicking one of them results in a error message with the text:
"Not found 404"
Error calling [/index.php/index.php/en/category/2/parts](http://192.168.1.216/index.php/index.php/en/category/2/parts).
Try to reload the page or contact the administrator if this error persists.

It looks to me as it is adding two instances of index.php in the call here. Not sure if i messed up my apache config or something like it? 

**To Reproduce**
Steps to reproduce the behavior:
Choose any of the categories in the menue

**Expected behavior**
Load it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
Part DB 1.10.2 
- PHP Version: 8.1.27
- Database Server 10.5.21-MariaDB-0+deb11u1 Debian 11

**Desktop (please complete the following information):**
 - Gentoo Linux 
 - firefox
 - Version 120

**Additional context**
Add any other context about the problem here.

RaptorDE, 2024-01-14:
Hi, I don't know if this is the solution, but I had a similar problem. For me it was that the part-db directories did not belong to the user and group www-data.

The error occurred when copying the part-db folders.

```
/var/www/html/uploads
/var/www/html/public/media
```

```
user@server:~/html$ ls -ls /var/www/html/public/media
total 12
4 drwxr-xr-x 5 www-data www-data 4096 Jan 14 17:52 cache
4 drwxr-xr-x 4 www-data www-data 4096 Jan 14 17:52 category
4 drwxr-xr-x 7 www-data www-data 4096 Jan 14 17:52 part
```

I fixed it with `chown -R www-data:www-data /var/www/html/public/media`.
It's best to check all copied directories.

ObligedTester, 2024-01-14:
Hello,

Thank you for your reply!

Turns out it was much more simple.

I thought this step was optional:

`sudo a2enmod rewrite`

But it was not, so after i run it everything works!

closed 2024-01-14 by ObligedTester