Part-DB/Part-DB-server #478: Browser crashing / not responding

splattner, 2024-01-14:
**Describe the bug**

When working with partdb my browsers (google and firefox) after a few minutes getting very slow up to unusable.
I also just got a "Error code: SIGILL" in Chrome. 
I have around (max) 50 parts

**To Reproduce**

I'm currently just working with partdb and adding some inventory. After a while, it feels unusable and slow. Firefox even messages me if I want to stop as its slow.

**Expected behavior**

Working without performance issues

**Server Side**
- Part-DB Version: 1.10.2 with Docker Image
- PHP Version: the one from your docker image I guess 8.1 based on the Dockerfile
- Database Server. I'm using the sqlite database
- Running it as a Pod on a Kubernetes Cluster. But I hat the same directly in Docker.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser Chrome (Version 120.0.6099.216 (Official Build) (64-bit)), Firefox 121.0 (64-bit)

jbtronics, 2024-01-14:
I have never encountered such issues. The only thing possible here is some flaw in the frontend code, which cause a very high memory usage. And even then, it should not be possible for a webpage to crash a browser.

Does this always happen, independently of what you are doing in Part-DB? or only after some special actions?

Do you encounter such issues also on the demo instance (https://https://demo.part-db.de/en)?

splattner, 2024-01-14:
Its way more severe when it loads all the parts (/en/parts page), I cannot load this page anymore. When in a subcategory with a few parts it seems better. Let me check if its from a specific category or part where the issue appears.

No, issues on the demo instance, so I guess something in my data..

splattner, 2024-01-14:
hmm.. I think this description `20V 420mV@1.1A 1.1A DO-219AB Schottky Barrier Diodes (SBD) ROHS` messed it up... somehow the `@` results in the stange behavior... but sure yet what and why exacly. The tables shows me

```
40V Single [&#38;&#35;&#x33;&#56;&#59;&#x26;&#35;&#120;&#x32;&#51;&#59;&#38;&#x23;&#x78;&#x37;&#56;&#x3b;&#x26;&#x23;&#x78;&#x33;&#x33;&#x3b;&#x26;&#x23;&#x34;&#x38;&#59;&#x26;&#35;&#120;&#x33;&#x62;&#x3b;&#x26;&#x23;&#120;&#x32;&#54;&#x3b;&#38;&#35;&#x78;&#x32;&#x33;&#59;&#x26;&#35;&#120;&#x37;&#x38;&#x3b;&#38;&#35;&#x35;&#48;&#x3b;&#38;&#x23;&#x31;&#x30;&#x31;&#x3b;&#38;&#x23;&#x35;&#x39;&#59;&#38;&#x23;&#51;&#56;&#59;&#x26;&#x23;&#x78;&#50;&#51;&#59;&#38;&#35;&#120;&#51;&#53;&#59;&#x26;&#x23;&#x78;&#x33;&#x33;&#x3b;&#x26;&#35;&#53;&#x39;&#59;&#38;&#35;&#120;&#50;&#54;&#x3b;&#x26;&#35;&#x33;&#x35;&#59;&#38;&#x23;&#53;&#x33;&#x3b;&#x26;&#x23;&#x35;&#49;&#x3b;&#x26;&#35;&#53;&#x39;&#x3b;&#x26;&#x23;&#120;&#50;&#54;&#59;&#38;&#35;&#51;&#53;&#x3b;&#38;&#x23;&#120;&#x33;&#56;&#59;&#x26;&#35;&#x78;&#51;&#54;&#x3b;&#x26;&#x23;&#53;&#57;&#x3b;&#x26;&#x23;&#120;&#x32;&#x36;&#59;&#38;&#35;&#51;&#53;&#59;&#x26;&#x23;&#49;&#x32;&#x30;&#x3b;&#38;&#x23;&#120;&#51;&#x34;&#x3b;&#x26;&#35;&#52;&#56;&#59;&#x26;&#35;&#53;&#x39;&#x3b;&#x26;&#35;&#120;&#x32;&#54;&#59;&#38;&#35;&#51;&#x35;&#59;&#38;&#x23;&#53;&#51;&#59;&#x26;&#x23;&#120;&#x33;&#x30;&#59;&#38;&#x23;&#53;&#x39;&#59;&#x26;&#35;&#120;&#50;&#54;&#x3b;&#38;&#35;&#120;&#x32;&#x33;&#x3b;&#38;&#35;&#53;&#x32;&#x3b;&#38;&#35;&#x35;&#52;&#59;&#38;&#35;&#120;&#x33;&#98;&#59;&#38;&#35;&#x33;&#56;&#x3b;&#x26;&#x23;&#x33;&#x35;&#x3b;&#38;&#x23;&#53;&#50;&#x3b;&#38;&#x23;&#53;&#x36;&#59;&#x26;&#x23;&#x35;&#57;&#x3b;&#x26;&#x23;&#x33;&#x38;&#59;&#x26;&#35;&#x33;&#53;&#59;&#38;&#x23;&#x31;&#x32;&#48;&#59;&#38;&#35;&#53;&#x32;&#x3b;&#x26;&#35;&#52;&#x39;&#x3b;&#x26;&#35;&#53;&#x39;&#59;](mailto:&#38;&#35;&#x33;&#56;&#59;&#x26;&#35;&#120;&#x32;&#51;&#59;&#38;&#x23;&#x78;&#x37;&#56;&#x3b;&#x26;&#x23;&#x78;&#x33;&#x33;&#x3b;&#x26;&#x23;&#x34;&#x38;&#59;&#x26;&#35;&#120;&#x33;&#x62;&#x3b;&#x26;&#x23;&#120;&#x32;&#54;&#x3b;&#38;&#35;&#x78;&#x32;&#x33;&#59;&#x26;&#35;&#120;&#x37;&#x38;&#x3b;&#38;&#35;&#x35;&#48;&#x3b;&#38;&#x23;&#x31;&#x30;&#x31;&#x3b;&#38;&#x23;&#x35;&#x39;&#59;&#38;&#x23;&#51;&#56;&#59;&#x26;&#x23;&#x78;&#50;&#51;&#59;&#38;&#35;&#120;&#51;&#53;&#59;&#x26;&#x23;&#x78;&#x33;&#x33;&#x3b;&#x26;&#35;&#53;&#x39;&#59;&#38;&#35;&#120;&#50;&#54;&#x3b;&#x26;&#35;&#x33;&#x35;&#59;&#38;&#x23;&#53;&#x33;&#x3b;&#x26;&#x23;&#x35;&#49;&#x3b;&#x26;&#35;&#53;&#x39;&#x3b;&#x26;&#x23;&#120;&#50;&#54;&#59;&#38;&#35;&#51;&#53;&#x3b;&#38;&#x23;&#120;&#x33;&#56;&#59;&#x26;&#35;&#x78;&#51;&#54;&#x3b;&#x26;&#x23;&#53;&#57;&#x3b;&#x26;&#x23;&#120;&#x32;&#x36;&#59;&#38;&#35;&#51;&#53;&#59;&#x26;&#x23;&#49;&#x32;&#x30;&#x3b;&#38;&#x23;&#120;&#51;&#x34;&#x3b;&#x26;&#35;&#52;&#56;&#59;&#x26;&#35;&#53;&#x39;&#x3b;&#x26;&#35;&#120;&#x32;&#54;&#59;&#38;&#35;&#51;&#x35;&#59;&#38;&#x23;&#53;&#51;&#59;&#x26;&#x23;&#120;&#x33;&#x30;&#59;&#38;&#x23;&#53;&#x39;&#59;&#x26;&#35;&#120;&#50;&#54;&#x3b;&#38;&#35;&#120;&#x32;&#x33;&#x3b;&#38;&#35;&#53;&#x32;&#x3b;&#38;&#35;&#x35;&#52;&#59;&#38;&#35;&#120;&#x33;&#98;&#59;&#38;&#35;&#x33;&#56;&#x3b;&#x26;&#x23;&#x33;&#x35;&#x3b;&#38;&#x23;&#53;&#50;&#x3b;&#38;&#x23;&#53;&#x36;&#59;&#x26;&#x23;&#x35;&#57;&#x3b;&#x26;&#x23;&#x33;&#x38;&#59;&#x26;&#35;&#x33;&#53;&#59;&#38;&#x23;&#x31;&#x32;&#48;&#59;&#38;&#35;&#53;&#x32;&#x3b;&#x26;&#35;&#52;&#x39;&#x3b;&#x26;&#35;&#53;&#x39;&#59;) 2A SMAF Schottky Barrier Diodes (SBD) ROHS
```

after saving the part

But when I type ``20V 420mV @ 1.1A 1.1A DO-219AB Schottky Barrier Diodes (SBD) ROHS` , so spaces around the `@` it seems ok....

jbtronics, 2024-01-15:
Interesting. I think this is related to marked-mangle, which basically tries to detect email addresses in markdown texts and mangle them, so they can not be scraped easily by bots. Therefore this issue only occurs when you have an email like format (no spaces around the `@`) and you end up with these crytic looking hex encoded text in HTML.

It seems that this plugin does not like it, if it gets registered multiple times at marked and somehow cause an heap overflow: (see this issue https://github.com/markedjs/marked-mangle/issues/126). Therefore this problem only occurs, after an page visit was already performed.

Thank you for reporting that. I will look into how to fix it properly. Maybe I will just remove the mangle plugin completly, as its not really required for most of Part-DB markdown usages.

splattner, 2024-01-15:
awesome, yeah it looks like that. Thx for investigating!

jbtronics, 2024-01-15:
This is fixed with Part-DB 1.10.3

closed 2024-01-15 by jbtronics