Part-DB/Part-DB-server #479: Docker, MySQL Backup failed, mysqldump: not found

RaptorDE, 2024-01-14:
I tried to backing up the MySQL database with the built-in backup function, but since mysqldump is missing in the Docker image v1.10.2, the backup fails.

This behavior is also mentioned in the [docs.](https://docs.part-db.de/usage/backup_restore.html#backup-automatic--part-db-supported)

Workaround: I installed mysqldump with docker exec.

```
docker exec partdb apt-get update
docker exec partdb apt-get install -y default-mysql-client
docker exec --user=www-data partdb php bin/console partdb:backup --full /var/www/html/public/media/backup.zip

```

Long term solution: Add mysqldump to docker image.

Or am I doing the backup wrong?

jbtronics, 2024-01-14:
From now on mysqldump should be included in the newly built docker images and the backup function should work properly.

RaptorDE, 2024-01-14:
Thanks for the quick answer, I'll test it with the next release.

closed 2024-01-14 by RaptorDE