Part-DB/Part-DB-server #483: KiCAD Integration - Showing only one part in each category

caiovernaglia, 2024-01-15:
**Describe the bug**
When I add my first part to a category it's shows up in Kicad. But the second part doesn't show.

**To Reproduce**
Steps to reproduce the behavior:
1. Add the first part and update Kicad (it's works fine)
2. Second part, on the same category, doesn't show up in Kicad
3. First part on another category, shows up, but the second one doesn't

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

[image]

**Server Side**
- Part-DB Version: 1.10.2
- KiCAD Version: 8.0.0-rc1.6

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser: Chrome
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jbtronics, 2024-01-15:
Try to close and open the part selector in KiCad or maybe even restart KiCad completely.
KiCad does some heavy caching, it is possible that it does not fetch new data from Part-DB yet and therfore you dont see the recent changes.

caiovernaglia, 2024-01-16:
No success. Not even a full PC reboot updated the library in KiCAD

caiovernaglia, 2024-01-16:
It requires more digging, but i'm assuming that's something related to KiCAD.

FYI: if I run a fresh install, add a bunch of parts and then import the KiCAD library, every parts are imported.
But, all changes after the initial import doesn't update.

jbtronics, 2024-01-16:
Does it change something if you clear Part-DB cache? (With `php bin/console cache:clear`)

caiovernaglia, 2024-01-16:
EDA_KICAD_CATEGORY_DEPTH=0

Running php bin/console cache:clear give me a warning
[image]

Continuing the command, Part-DB became broke.
[image]

On the first screenshot, it's says "delete the var/cache directory completely and let it be recreated by Part-DB.
No success.

jbtronics, 2024-01-16:
You must run the console as the webserver user (e.g. something like `sudo -u www-data php bin/console`, otherwise the cache files get the wrong permissions, and you get this 500 error.

Delete the `var/cache` folder in Part-DB completely and refresh the Part-DB page.

caiovernaglia, 2024-01-16:
www-data
This account is currently not available.

Sorry for inconvenience, but i'm not a Linux expert. I understand a few commands only.

caiovernaglia, 2024-01-16:
Another comment, I've tried to Wireshark communication between KiCAD and PART-DB, and I could confirm that PART-DB kicad-api is only replying with one part in each category.
[image]

So, it's looks like to be some cache on PART-DB Kicad-API.

Normal API is respnding with all parts

jbtronics, 2024-01-16:
Okay, I will look into that.

caiovernaglia, 2024-01-16:
Correction for my last comment.

It's not only one part. PART-DB Kicad-API is always replying with the same payload as the first request from KiCAD.
ie:
Adding multiples parts to PART-DB and then add the library in KiCAD, the first request gets all parts.
All others request, KiCAD receives the same payload, even with somethings changes in PART-DB.

caiovernaglia, 2024-01-16:
> Okay, I will look into that.

Thanks!

BTW: Excellent job!!!

jbtronics, 2024-01-16:
There actually was a bug in Part-DB that the cached responses were not properly invalidated and therefore changes were not shown.

This should be now fixed in Part-DB 1.10.4

And always keep in mind, that only parts which have meaningful data (so therfore a symbol and/or footprint defined) or are explicitly marked as visible, show up in KiCAD.

caiovernaglia, 2024-01-16:
Ohh, nice!

1.10.4 is already in docker-hub?

jbtronics, 2024-01-16:
Yes now it should. Docker images build always take a while, as the cross builds for ARM are very slow.

https://github.com/Part-DB/Part-DB-server/actions/runs/7548322919/job/20550134790