Part-DB/Part-DB-server #484: [1.10.3 (docker)] Unable to create or edit categories

SANCLA, 2024-01-16:
**Describe the bug**
Docker pulled a new part-db version last night.
Since then, I cannot create new or edit existing categories.
Input fields are grayed out.

**To Reproduce**
1. Go to tools -> edit -> categories
2. Click on new element
3. input fields are grayed-out

**Expected behavior**
Being able to create a part with provided information

**Server Side**
- Version: 1.10.3 (docker)
- MariaDB SQL database (separate docker instance)

**Desktop (please complete the following information):**
 - OS: macOS 14.2.1 (23C71)
 - Browser: Firefox 121.0

jbtronics, 2024-01-16:
Are you logged in as a user with edit rights? Maybe you need to refresh the browser to see that you are logged out.

SANCLA, 2024-01-16:
Owwwww *slapping myself on the forehead*
Never mind...

closed 2024-01-16 by SANCLA