Part-DB/Part-DB-server #487: Scanning Digikey Datamatrix Codes -  Format unknown

stuartmckim, 2024-01-19:
**Describe the bug**
When using the label scanner to scan the datamatrix code on Digikey bag labels, Part-DB always returns the error "Format unknown!" in red at the top of the screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Open label scanner.
2. Scan a Digikey datamatrix code.

**Expected behavior**
The Digikey datamatrix code gets used to get information on the part.

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.10.4
- PHP Version: 8.2.7
- Database Server sqlite3

**Desktop (please complete the following information):**
 - OS: Linux (Debian)
 - Browser: Firefox
 - Version: 115.6

**Smartphone (please complete the following information):**
 - Device: Pixel 2
 - OS: Android
 - Browser: Firefox
 - Version: 121.1

**Additional context**
The datamatrix is scanned, and the contents dumped into the Input field, but Part-DB doesn't seem to know what to do with that information. Behavior is same on desktop and smartphone.
Adding the part first, by way of the Digikey info provider, makes no difference.
What I'm trying to do: scan my parts as they come in for an order to then create new parts in Part-DB. I'm assuming that kind of thing was the intent of scanning vendor barcodes.

jbtronics, 2024-01-19:
Part-DB can not automatically detect vendor barcodes, and I guess thats pretty much impossible, as they should be unique for every lot.
To make this feature work, you need to go to a parts edit page, select the lot/stock, with which you want to associate this datamatrix code and input the content of the barcode (so what the scanner reads from the datamatrix code) in the "vendor barcode" field under advanced settings:
[image]

If you then scan the datamatrix code again, it should open the page of the part.

stuartmckim, 2024-01-19:
Ah, OK. That makes sense. I guess I was hoping it would extract the part number from the data, but I can see how that level of detail is useful for some people. Since it will change with each order, and I don't need lot-level tracking, I'll skip it.
Thank you for your quick help.

closed 2024-01-19 by stuartmckim

aWZHY0yQH81uOYvH, 2024-05-01:
I agree it would be very useful to be able to scan DigiKey labels, have their part information queried from the API, and bring up the "add part" screen if the part is not already in the database. Although the full content of the barcode is different for each order, the manufacturer part number [is in there in a known location](https://hackaday.io/project/90456-ar-workbench/log/131388-reading-digikey-barcodes).