Part-DB/Part-DB-server #489: ZPL/EPL Support for direct label printing.

fiz-o-matic, 2024-01-22:
Hi,

It's possible to add the capability to print labels direct to a label printer via ZPL or EPL?
Many professional label printer use a special language to print direct a label from the application without a print server.
Brother and Zebra printer supporting ZPL and EPL to print a labels.

Thanks!

jbtronics, 2024-01-22:
Part-DB runs in a browser and must be able to communicate with the printer directly. the normal browser APIs dont support that, so you most likely require a browser plugin or companion app to do that.

The bigger problem is that Part-DB labels are HTML, which are rendered into PDF. It seems that ZPL is pretty limited in terms of text formatting compared to HTML/PDF. So you cannot really convert the labels into ZPL, without big changes. I think the best possible, would be to convert the labels into an image and then let the printer print that image on the labels.

I would assume that label printers should have some kind of printer driver, which allows them to print arbitary stuff via the windows printer subsystem. And in this case you can print the labels from Part-DB from inside the browser with the print dialog of windows/your browser.

fiz-o-matic, 2024-01-22:
ZPL is designed to print the label directly from the application or from the server.
So you don't need a special driver or a special subsystem for printing.

I think there are some php libraries to communicate with a printer directly over ethernet (or USB).

At the moment I have to download the generated PDF and print it over CUPS.
On my OSX I can't print directly from the page because I can't see anything in the preview window.

jaimelaborda, 2024-03-02:
I also love this. I iam currently planning on doing some applicatio

whc2001, 2024-05-07:
I also need something like this. The printer I use only supports TSPL over USB. My old solution with PartKeepr was to made a local application to packet capture the browser requests, on each new part submission it grabs some of the information and construct the TSPL script by hand then send to the printer raw device. I wonder if the same could work for Part-DB...

btoplak, 2025-02-14:
Hi all.

QZ Tray is one of the prospect solutions, it enables browser-to-hardware raw printing with support for multiple programming languages like ZPL, EPL, ESC/POS etc, so you can print on most of the label printers and also POS temal printers...

Documentation is good, implementation into this app should be a breeze...

More here
- https://github.com/qzind/tray
- https://qz.io/
- DEMO: https://demo.qz.io/

z3ugma, 2026-09-10:
Could use Web Serial tech? or Web USB? for the browser app to talk directly to the printer over USB or Serial