Part-DB/Part-DB-server #494: API docs wrong auth header

Flo2410, 2024-01-25:
**Describe the bug**
In the API docs it states that the auth header is `Authentication`, but in reality it is `Authorization`.

> Note: Domain and API token are redacted.

**To Reproduce**

The request
```bash
curl -X 'GET' 'https://partdb.example.com/api/info' -H 'accept: application/ld+json' -H 'Authentication: Bearer some_api_token'
```

gets the response:
```html
<!DOCTYPE html>
<html>
    <head>
        <meta charset="UTF-8" />
        <meta http-equiv="refresh" content="0;url='http://partdb.example.com/en/login'" />

        <title>Redirecting to http://partdb.example.com/en/login</title>
    </head>
    <body>
        Redirecting to <a href="https://partdb.example.com/en/login">http://partdb.example.com/en/login</a>.
    </body>
</html> 
```

**Expected behavior**

The correct way is:
```bash
curl -X 'GET' 'https://partdb.example.com/api/info' -H 'accept: application/ld+json' -H 'Authorization: Bearer some_api_token'
```
This returns the expected JSON response. 

Also, shouldn't the API respond with an `401 Unauthorized` error instead of redirecting to the login page?

**Server Side**
- Part-DB Version: 1.10.4
- PHP Version: 8.1.27 (SAPI: fpm-fcgi)
- Database Server: 10.4.12-MariaDB

**Desktop (please complete the following information):**
 - OS: Debian 6.5.13-1 (2023-11-29) x86_64 GNU/Linux
 - Kernel: Linux 6.5.0-5-amd64
 - Shell: zsh 5.9 (x86_64-debian-linux-gnu)
 - curl: 8.5.0 (x86_64-pc-linux-gnu)

jbtronics, 2024-01-25:
Thanks for the notice. I corrected the header name in documentation

jbtronics, 2024-01-25:
With the latest version from the master branch, a 401 response with an JSON encoded error message, should be returned, when the authentication header is missing.

closed 2024-01-25 by jbtronics