Part-DB/Part-DB-server #503: Mouser API is only returning part image, NO Parameters or Datasheet link

clytle374, 2024-01-27:
I just got partsDB installed on Gentoo Linux with Docker.  This was all new territory for me.

I installed Digikey API and spent days inputting data, then Octopart, now I'm trying Mouser.

When I search the part I get the results, but when I click add part I have a blank parameters page.  I do get the attachment populated for product image, but no datasheet.  Looking on the Mouser site it looks like the API should be giving this information like Digikey and Octopart did.  

Is this user error, limitation, or a bug?

 A couple other things.  I don't have a bin/console like called for in the instructions.  And it took a while to figure out that `docker-compose` was actually `docker compose` on my system.

clytle374, 2024-01-27:
More information, have those lines enabled in docker-compose.yaml causes digikey search > add part to return and internal server error.  

```

      - PROVIDER_DIGIKEY_CLIENT_ID=
      - PROVIDER_DIGIKEY_SECRET=
      # The currency to get prices in
      - PROVIDER_DIGIKEY_CURRENCY=USD
      # The language to get results in (en, de, fr, it, es, zh, ja, ko)
      - PROVIDER_DIGIKEY_LANGUAGE=en
      # The country to get results for
      - PROVIDER_DIGIKEY_COUNTRY=US

#      - PROVIDER_OCTOPART_CLIENT_ID=
#      - PROVIDER_OCTOPART_SECRET=
#      - PROVIDER_MOUSER_KEY=
```
Uncommenting the last 3 lines causes the add part fault.  I deleted all the private information.

jbtronics, 2024-01-27:
Part-DB parses (almost) every information returned by the Mouser API. However it seem that it is broken in that aspects:

There is a `DataSheetUrl` field in the API response, which Part-DB would use to create datasheets. However the Mouser API returns always an blank value.

Similar for the  `ProductAttributes` field. Instead of the parts specifications shown on the mouser page, it just contains two entries about the packaging.

As Mouser does not provide this information via its API, Part-DB can not retrieve them.

I will add at least a hint to the docs, that this behavior is denoted.

jbtronics, 2024-01-27:
> Uncommenting the last 3 lines causes the add part fault. I deleted all the private information.

On my test system it is running fine, with all providers enabled simultaneously. Ensure that all authentication tokens for the APIs are correct and  try to clear Part-DBs cache.

clytle374, 2024-01-28:
> > Uncommenting the last 3 lines causes the add part fault. I deleted all the private information.
> 
> On my test system it is running fine, with all providers enabled simultaneously. Ensure that all authentication tokens for the APIs are correct and try to clear Part-DBs cache.

I will try this.  When all 3 lines were on I could use mouser and octopart.  I installed it on a 2nd computer overnight to be sure my backup tarball of the folder was would work. I wanted to be sure before doing anything else as entering the parts is not something I want to repeat.    

Many of the commands include bin/console  I cannot find this on my system.  The following command does work and it includes bin/console.  

  `docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate`

This one does not `bin/console cache:clear`  Again, I apologize as I really have no idea what I'm doing with php, docker, or container.  In order to get the change to take effect I used the commands to update.

jbtronics, 2024-01-28:
> Many of the commands include bin/console I cannot find this on my system. The following command does work and it includes bin/console.
> 
> `docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate`

yes if you use docker, you have to replace run the console commands that way. But its basically the same. Just replace the part after bin/console with the command you want to exectute.

Therfore if you want to clear the cache, you will need to do something like:

`docker exec --user=www-data partdb php bin/console cache:clear`

clytle374, 2024-01-29:
That makes since once somebody said it.  Thanks.  
In order for changes to docker-compose.yaml to take effect I ran

> docker compose up -d

Is this the correct action, might be helpful to add that to the docs?  I don't think I was supposed to put the info provider info in this file, but the others didn't work.  But I knew even less about what I was doing then.

Problem is resolved, thanks again

d-buchmann, 2024-01-31:
Just adding one data point:
Retrieving datasheets through the Mouser info provider generally works (try e.g. ATmega169PA-AUR).
I suspect the problem is that Mouser sometimes links to datasheets on their domain, which they seem to supply through the API - but not to external sites, which they do on their website though.

closed 2024-02-24 by jbtronics