Part-DB/Part-DB-server #504: SAML users unable to use site but able to login

msessions61, 2024-01-29:
**Describe the bug**
SAML users are able to login and are auto assigned roles based on keycloak groups with no issue. Upon login they can only access the home page and the user info page. SAML users receive a 500 error when attempting to add new parts, go to the user settings page or do anything beyond the 2 functioning pages mentioned. On the user info page it shows their assigned role and that they should have all permissions available to admins.

Local users (users that login via username and password) are able to login the site and navigate, add parts, adjust site settings and more with no issue. New local users can be added and modified with no issue.

**Expected behavior**
For SAML users to be able to use the site to the level that their permissions allow for.

**Screenshots**
[image]
[image]
[image]

**Server Side**
- Part-DB Version: 1.10.4
- PHP Version: 7.4.20
- Database Server MySQL 5.1.1

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 120.0.6099.227

**Additional context**
This instance of Part DB is being run in a docker container

jbtronics, 2024-01-29:
What does the server logs say? They should be available via `docker logs` ?

jbtronics, 2024-01-29:
The bug should be fixed in the new 1.10.6 release

msessions61, 2024-01-29:
Seems to be working!

closed 2024-01-29 by msessions61