Part-DB/Part-DB-server #508: lighttpd support

jon-daemon, 2024-02-04:
Is it possible to include a conf file for lighttpd server?
I already have lighttpd installed for pihole on a device with very low resources and I'd like to use that for part-db too.

jbtronics, 2024-02-04:
If lighttpd is already configured to use PHP, it should not be too difficult to use Part-DB with it. Basically you just need to add a new vhost definition in lighttpd, pointing the the correct document root and adding the required rewrite routes.

Something like this should work

```
$HTTP["host"] =~ "myhost\.tld" { 
    server.document-root = "/var/www/partdb/public" 
    accesslog.filename = "/var/log/lighttpd/partdb_access.log;" 
    url.rewrite-if-not-file = ( 
        "^/$" => "$0", 
        "^/(?!.+\.php)[^\?]+(\?.*)?" => "index.php/$1$2", 
    ) 
}

```

gstrauss, 2024-02-07:
@jbtronics I think you made some typos in the above, such as ';' at end of partdb_access.log.  I think you meant
```
$HTTP["host"] == "myhost.tld" {
    server.document-root = "/var/www/partdb/public"
    accesslog.filename = "/var/log/lighttpd/partdb_access.log"
    url.rewrite-if-not-file = (
        "^/(?!.+\.php)[^\?]+" => "/index.php/${url.path}${qsa}",
    )
}
```
https://wiki.lighttpd.net/mod_rewrite

jbtronics, 2024-02-07:
Thats possible. I basically copied this config snippet and modified it slightly on phone, so its possible that it contains some typos. I have not tested it. So you should give it try and give feedback if it work with that config.

gstrauss, 2024-02-08:
From where did you copy that config snippet?  I would like to get that corrected.  Thanks.

jbtronics, 2024-02-08:
Not from the official lighthttpd documentation. I think it was from another open source project who uses symfony. It was used for an old symfony version though, so i had to slightly modify it.

In general it seems that running a symfony application with lighttpd is not so common. I were not able to find much info about that (even though i would think that the correct PHP and rewrite config should be enought)