Part-DB/Part-DB-server #515: Search in parts does not work

henrikkoch, 2024-02-11:
I have just created a lot of parts from DigiKey

F.ex. this part number 691137710004‎

I was successfully looking up the data from digikey and created in my list of parts (PartDB)

Afterwards I search for the number I the upper right corner and the part cannot be seen even though its in my list

[image]

jbtronics, 2024-02-18:
For some reason I am not able to replicate the bug anymore. It seems to work fine on the demo instance: https://demo.part-db.de/en/parts/search?name=1&category=1&description=1&mpn=1&tags=1&storelocation=1&comment=1&ipn=1&ordernr=1&keyword=691137710004

Can you please recheck if this still happens with the recent versions?

henrikkoch, 2024-02-18:
I maybe have some more info. It looks like the bug is there right after the
new component has been created.

Just after import (creation) of new part
Search for the same part number

After some time my problem also seams to have disappeared. Maybe just some
refresh of the data right after import is the real issue here,

Den søn. 18. feb. 2024 kl. 22.31 skrev Jan Böhmer ***@***.***

>:

> For some reason I am not able to replicate the bug anymore. It seems to
> work fine on the demo instance: [...]

jbtronics, 2024-02-18:
> F.ex. this part number 691137710004‎

Have you copied and pasted this number from your search attempts?

Because it contains a non-visible control code (0x200E, Left-To-Right Marker), which prevents the search from being successful. If you remove it (by typing backspace), then search works fine for me.

henrikkoch, 2024-02-18:
I created this:   "1N5408G_R2_00001" from TME

Right after import I searched for "1N5408G_R2_00001" and this was not found
in the 1.10

But again I believe it is something that is not refreshed

/Henrik

Den søn. 18. feb. 2024 kl. 22.31 skrev Jan Böhmer ***@***.***

>:

> For some reason I am not able to replicate the bug anymore. It seems to
> work fine on the demo instance: [...]

jbtronics, 2024-02-18:
> I created this: "1N5408G_R2_00001" from TME Right after import I searched for "1N5408G_R2_00001" and this was not found in the 1.10

I tried this and for me search shows the correct result even directly after creation.

> But again I believe it is something that is not refreshed

There is nothing which would require refreshment. Only the categories, locations, etc. for the sidebar trees are cached. The part search works directly on the database, and therefore should always show the current database entries, And the new part is directly inserted into the database in the save request. So directly after you see the "Save successfull" flash message, the part should show up in the search.

I suspect that you somehow end up inserting an invisible control code into your search term, which prevents the search from being successfull (as the part name does not contain this special character).

closed 2024-03-01 by jbtronics