Part-DB/Part-DB-server #518: Selections in sidebar are continuously lost

RobThree, 2024-02-13:
**Describe the bug**
Any selection I make in the left sidebar is immediately lost when the page is loaded (or if you happen to accidentally miss-click the +).

**To Reproduce**
Steps to reproduce the behavior:
1. Expand a few nodes in the left sidebar, preferably a few levels nested
2. Click the desired node
3. Page loads
4. Tree collapses, node is no longer visible

**Expected behavior**
The tree should stay as-is.

**Server Side**
- Part-DB Version: 1.10.7 (master/43d8c3c)
- PHP Version: 8.2.8 (SAPI: fpm-fcgi)
- Database Server: 10.11.6-MariaDB-0+deb12u1

**Desktop (please complete the following information):**
 - OS: macOS (but also on Windows 11)
 - Browser Chrome
 - Version: latest

**Additional context**
I have pretty deeply nested storage locations for example; in essence it's: `Home -> Cupboard -> Assortment cabinet -> Row -> Col -> Index`. When I try select `Home -> K3 -> F -> 7 -> b -> 1` and I, god forbid, happen to miss clicking _exactly_ on the + symbol, everything collapses. And when I do reach the desired node and would like to check another drawer in row 7 I need to expand everything up to that again.

jbtronics, 2024-02-18:
Does this also happens on the demo instance for you? Because I cannot reproduce the issue on my test setup, even for such deep  structures.

It sounds like somehow a full page reload is triggered, instead of the single page request. Have you tried to rebuild the frontend with `yarn build` and clear Part-DB cache afterwards `php bin/console cache:clear` (everything runned as the webserver user).

RobThree, 2024-02-18:
That fixed it! I ran `yarn build` and `php bin/console cache:clear` and it was fixed. Thanks!

closed 2024-02-18 by RobThree