Part-DB/Part-DB-server #520: Creating new parts from vendor formats (for example with a scanner)

meganukebmp, 2024-02-19:
Hello, I currently have a lot of parts from vendors that supply json formatted data matrix codes on the received part (as well as order sheet). This json provides part name, quantity as well as description. This is generally really useful if I want to bulk add new parts, which sometimes happens. For this usecase I had made a simple program interpreting the code and communicating with the API to import these codes. However that in my opinion is quite a lot of work, especially if working with many different vendors.

The feature proposal is to have some sort of automated, scanner focused, part creation functionality in the UI level with custom format definitions. This could for example work as follows:
* User selects format type (JSON, XML, Other?)
* User matches fields from the format to fields in the part (for example {"name": "partname", "data": {"quantity":2, "resistance": 470}} could then be field matched by inputting "name" as the value for the name field and "data.quantity" for stocks fields)
* The user goes to a add new part dialog and scans a vendor code (pre-selected vendor code type since auto detection might be a lot of work)
* The part gets created according to the format rules

A system like this would be very versatile and would save a lot of time when having to import hundreds of components when only their physical labels are available.

jbtronics, 2024-02-20:
So you basically want something like the existing part import from JSON files, but from a scanned matrix code?

Im curious, what vendor is this? Sounds like an odd way to provide part data to customers.

meganukebmp, 2024-02-22:
> Im curious, what vendor is this? Sounds like an odd way to provide part data to customers.

LCSC. Here's an example raw scan of their QR codes:
```
{pbn:PICK2311060022,on:GB2311060577,pc:C3001172,pm:CH32V203C8T6,qty:10,mc:,cc:1,pdi:95798469,hp:0,wc:ZH}
```

codeman256, 2024-03-04:
I just finished scanning a bunch of Digikey products we had lying around into our instance. I used a form that splits the DataMatrix code up from

[)>06PTEMPERATURE SENSOR1PMCP9700A-E/TOK1K4962xxxx10K5687xxxx11K14LQ311ZPICK12Z121250813Z11085820Z00000....(many zeros)

 - those are GS codes separating the values

to this. 

| Fieldname | code | value |
|---|---|---|
| Part Num else DKN | P | TEMPERATURE SENSOR |
| MFG PN | 1P | MCP9700A-E/TO |
| PO | K |  |
| Salesorder ID | 1K | 4962xxxx |
| Invoice ID | 10K | 5687xxxx |
| Picklist ID | 11K | 1 |
| COO | 4L |  |
| Quantity | Q | 3 |

then used part-db's API to insert the part and the part-lot(QTY) into part-db. took way longer than it should have but it's done now. 
Also found this online for reference.
[image]
https://forum.digikey.com/t/digi-key-barcodes/7332

I still had to go into partdb and still use the find part from provider feature to pull down all the part data but at least there was less typing. idk haha

d-buchmann, 2024-06-26:
FYI, Würth Elektronik, ON semiconductors and probably others use a similar format (to Digikey) on their components as well.
I did some research and traced the common ancestor through ISO/IEC 15434 / ISO/IEC 15418 -> MIL-STD-130N -> ANSI MH 10.8.7 -> CEA-706.
Interestingly, the MIL standard is the only one [freely available](https://www.mil-std-130.com/media/pdf/specs/mil-std-130n.pdf).
The most difficult task for this format would be to find/configure scanning hardware so that it correctly reads and propagates the special symbols; otherwise, that format is close to useless.
Another option is to have an extra local software to do the scanning and conversion like @codeman256 suggested; ideally to a "normalized" JSON that Part-DB can import directly.

However, I think this is most useful for vendors/manufacturers that we don't have an info provider for. For Digikey and LCSC, we can use only the MPN and their respective API to retrieve the latest information (including pricing) - except you are interested in order specific info.

liborw, 2025-03-10:
Here is TME example:
```
QTY:10 PN:T491A226M006AT PO:32075995/10 MFR:KEMET MPN:T491A226M006AT CoO:MX RoHS https://www.tme.eu/details/T491A226M006AT
```

swdee, 2026-01-18:
I have made a PR that implements making new parts with the Scanner for LCSC, Mouser, and Digikey labels.   We require users to test out this feature and report back any bugs.    Please visit the PR #1194 and watch the Demo videos, then try out with your own devices on the [test server](https://github.com/Part-DB/Part-DB-server/pull/1194#issuecomment-3764634434).

jbtronics, 2026-02-25:
Implemented with PR #1194

closed 2026-02-25 by jbtronics