Part-DB/Part-DB-server #522: Migration error

BlackDrakon, 2024-02-20:
**Describe the bug**
We are using PartDB version 1.8.2 (last 1.8). After upgrade to any (we tried v1.9.0 v1.9.1 v1.10.0 v1.10.7) newer version it stops working. All migrations and updates are goes well but after upgrade page can not retrieve components list. Structure of storage is observable. Users are there. Clear cache update upgrade migrate doesn't help. Logs doesn't says anything what can lead to the problem.

Attachments also observable. But any klick on elements list or storage give error 404. Elements also can be shown with direct link.

**Screenshots**
[image]
[image]
[image]

**Server Side**
- Part-DB Version: 1.8.2 to 1.9.0
- PHP Version: 8.1 - 8.3
- Database Server: 10.6.16-MariaDB-0ubuntu0.22.04.1

**Desktop (please complete the following information):**
 - OS: windows
 - Browser chrome

jbtronics, 2024-02-20:
That is not a database problem, but it seems to be some URL rewrite issue (as the generated path contains index.php twice).

You should try to setup your webserver, that URL rewrite works properly and you can use URLs like `/en/` instead of `index.php/en/`. See the installation documentation on how to configure Apache and nginx for that.

In principle it should also work without the  rewriting, but that is not really well tested an might lead to weird behavior like the one you observe. 

Have you set the `NO_URL_REWRITE_AVAILABLE` config option to true?

Also you should maybe try to upgrade to Part-DB 1.10.7 directly, maybe some stuff was already fixed there. And you can skip versions without any problem (just run `doctrine:migrations:migrate` to update the database afterwards).

BlackDrakon, 2024-02-20:
I have no idea why. But this time after update to version v1.10.7 it works fine without any problems...

closed 2024-02-20 by BlackDrakon