Part-DB/Part-DB-server #523: Required fields when adding a part

influx3k, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
Sometimes people forget to add the Storage Location (or other fields) in when adding a part in.

**Describe the solution you'd like**
If you could set required fields when adding a part, you could then make it so that the Storage Location (or any other field) has to be set in order to even add the part.

**Additional context**
Would love to be able to set required fields, for any field(s) of your choosing.  This would be super useful.  For me, I would be able to make the Storage Location a required field, so that users can't forget to add it and lead to a situation where we have an entry, but no location.  Just happened today or two items and we spent a considerable amount of time hunting around for them.  I added this is the Wiki section, but has no traction there.  Thank you.

NikoDelarich, 2024-08-22:
Seconded! I think it would be important to allow defining "required" fields or parameters - probably per category.
For instance, I would like to define that all new parts in the "Resistors"-category must have a "Resistance" parameter defined.

influx3k, 2024-11-26:
I cannot express how much a feature like this is needed.  Despite multiple reminders, people just can't remember to put in the storage location.  If it was required, we would not have this issue.  Please consider implementing something to address this.