Part-DB/Part-DB-server #531: May I ask, what is the relationship between PartKeepr and part-db? PartKeepr和part-db是什么关系？

stuffren, 2024-02-26:
May I ask, what is the relationship between PartKeepr and part-db?
Is it also developed by your team?
I see that PartKeepr supports CSV and Excel export and CSV import, does part-db support this feature?
请问，PartKeepr和part-db是什么关系？
它也是你们团队开发的吗？
我看到PartKeepr支持CSV和Excel导出以及CSV导入，part-db支持这个功能吗？

jbtronics, 2024-02-26:
Part-DB and PartKeepr are independant projects developed by different developers. However both projects try to create a good inventory managment software for electronics components. PartKeepr has not received much updates for a long time, and Part-DB in this repository is generally a bit more modern and newer. Part-DB can import existing PartKeepr databases to help users, who want to switch from PartKeepr to Part-DB.

Part-DB can import and export parts and data structures as CSV and other formats. Working with excel files directly is not supported, but excel and libreoffice should be able to convert CSV into an xlsx file and vice versa.

stuffren, 2024-02-26:
You say that Part-DB can import and export parts and data structures in CSV and other formats. I just tried to look for this feature in the demo site but couldn't find it, where is it please?
I was just planning to import data to Part-DB via csv, but of course the csv format must be according to the template of the corresponding field of the website.

In addition, when I saw this project of yours, I feel very surprised and would like to make friends with you, looking forward to communicate with you privately and deeply.

jbtronics, 2024-02-26:
You can export parts, by mass selecting them in the table, and choose "Export to CSV" in the appearing menu.

Import/Export of data structures like categories, footprints, etc. is possible on their admin pages: https://demo.part-db.de/en/category/new#import_export

For importing parts there is a own page. However all import functions are disabled in the demo instance, to prevent abuse. Therfore you can not see it

stuffren, 2024-02-26:
1.
I tried and this import button is not available (no response after clicking it).
This import button is supposed to import category information via csv, right?
The export button is available, but when I open the CSV file, the data is not organized by fields.

2.
Also is there a function that can directly import the model parameters directly under the category?
For example, I want to import 125 sub-models of Atmega 328 via CSV.

3.
Is there a separate backend management system for this system? Or as long as the administrator logs in through the top right corner of the home page, he/she can operate and manage the data.
can operate and manage the data. But ordinary users (non-administrator users) can only browse and view the data.
They can't do anything with the data.
Am I understanding this correctly?

jbtronics, 2024-02-27:
> I tried and this import button is not available (no response after clicking it).
> This import button is supposed to import category information via csv, right?

Yes thats right, because the import functions are all disabled in the demo to prevent misuse.

> Also is there a function that can directly import the model parameters directly under the category?
> For example, I want to import 125 sub-models of Atmega 328 via CSV.

For basic fields like "name", "category", "description", etc. this should be pretty easy. For stuff like real part parameters, which can have arbitary names, etc. you need to get the right file structure. Maybe something like JSON is easier for your purpose.

Also Part-DB allows you to pull part informations directly from distributors like LCSC or digikey. Maybe thats easier for you than importing.

> Is there a separate backend management system for this system? Or as long as the administrator logs in through the top right corner of the home page, he/she can operate and manage the data.
> can [...]

There is no seperate backend to manage Part-DB. Everything is possible through the unified interface. However you can control via permissions, what users can do and see. So what normal users or not-logged in users can do is completly your decision and you can control that pretty detailed.

stuffren, 2024-02-27:
Thanks for your reply. It gave me further insight into Part-DB.
I have some good ideas and would like to communicate with you further, let me confirm your e-mail address is
mail@jan-boehmer.de?

jbtronics, 2024-02-27:
yes

stuffren, 2024-02-27:
> yes

hello,[jbtronics](https://github.com/jbtronics), I've sent you an e-mail. Please check it.

closed 2024-03-01 by jbtronics