Part-DB/Part-DB-server #534: Installing the latest Git revision results in a Yarn error

frank-f, 2024-03-02:
**Describe the bug**
Installation currently ends with the following error:

```
ERROR  Failed to compile with 2 errors                                                                                                                                                          11:22:16 AM

Module build failed: Module not found:
"./node_modules/@symfony/stimulus-bridge/lazy-controller-loader.js!./assets/controllers/elements/part_search_controller.js" contains a reference to the file "@algolia/autocomplete-js".
This file can not be found, please check it for typos or update it if the file got moved.

"./node_modules/@symfony/stimulus-bridge/lazy-controller-loader.js!./assets/controllers/elements/part_search_controller.js" contains a reference to the file "@algolia/autocomplete-plugin-recent-searches".
This file can not be found, please check it for typos or update it if the file got moved.

webpack compiled with 2 errors
error Command failed with exit code 1.
```

**To Reproduce**
Steps to reproduce the behavior:
1. `git pull`
2. `composer install --no-dev -o`
3. `yarn build`

**Expected behavior**
I don't recall exactly what the output was previously, but I don't remember seeing a red ERROR.
My PartDB Installation seems to work fine like this, but that may not be the case for all setups?

**Server Side**
- Part-DB Version: f5d53dc  (HEAD)
- PHP Version: 8.1.27
- Database Server: MySQL 8.0.35
- Yarn: v1.22.19

jbtronics, 2024-03-02:
Have you run yarn install, to pull the new dependencies?

frank-f, 2024-03-02:
Oops, how could I miss that one. I don't recall ever running that (except maybe the initial installation), but somehow this was never a problem during previous updates.

Anyway, thanks for the quick reply - it works fine now. 👍

closed 2024-03-02 by frank-f