Part-DB/Part-DB-server #538: Create new part doesn´t  like categories with commas

nomike, 2024-03-04:
**Describe the bug**
When creating a new part from a provider (digkey in this case), the "Category" field is set to the first part in the list.
When I forget to change that, it is put to that category.

Below that field it prints the provider category information:

```
Provider: Crystals, Oscillators, Resonators -> Crystals
```

If I copy/paste this to the category field, part-db would create the following categories:

```
Resonators
  Crystals
```

**To Reproduce**
Steps to reproduce the behavior:
1. Try to create a part from the DigiKey provider

**Expected behavior**
I would expect the following categories to be created:

```
Crystals, Oscillators, Resonators
  Crystals
```

Things could be further improved by suggesting to create that category per default, instead of using the first one in the list. Or there could at least be a button which, using client side java script or something) copies the provider's category to the field.

**Server Side**
- Part-DB Version: 1.10.7
- PHP Version: 8.2.7
- Database Server: sqlite 3.40.1

**Desktop (please complete the following information):**
 - OS: Ubuntu 23.10
 - Browser: Firefox
 - Version: 123.0

closed 2024-03-04 by jbtronics

nomike, 2024-03-05:
I've updated to Part-DB 1.11.1 but the issue still seems to be there.

[image]

It creates these categories:

[image]

Which first of all is back at that ',' issue and secondly completely ignores that these two new categories should be descendants of "Connectors, Interconnects".

jbtronics, 2024-03-05:
Have you rebuild your frontend dependencies after upgrade and perform a cache clear in Part-DB and your browser?
Part of the fix was a frontend thing, that the input checkbox does not split up input on commas, so your browser needs to load the fresh assets.

In the demo instance that it works fine for me.