Part-DB/Part-DB-server #540: Mouser price import faulty?

Tropaion, 2024-03-04:
**Describe the bug**
I just tried the import function from mouser and created an Search-API for it.
Now I tried to update a current part, but the prices of the mouser part are wrong.
It also says that the part was announced even though it clearly is released and in stock

**To Reproduce**
Try importing this part to partdb: https://www.mouser.at/ProductDetail/Littelfuse/65600201009
It says announced:
[image]
Prices all zero:
[image]
If the search already says anounced, shouldn't it not try to add prices?

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: 1.11.0
- PHP Version: 8.1.27<html>
<body>

<html>
<body>

--

</body>
</html>
--

</body>
</html>
- Database Server [e.g. MySQL 5.7]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jbtronics, 2024-03-04:
The mouser api returns prices as locale dependent string depending on the selected country. Austrian mouser returned some string different to the expected one. This should be fixed with 1.11.1

Before, all parts which were marked as "New" by mouser were marked as "Announced". Now this only happens, if the part is new and no parts are instocked at mouser yet (otherwise its just active).

closed 2024-03-04 by jbtronics

Tropaion, 2024-03-04:
@jbtronics thanks, that was fast. Tried it just now, but accidently found another bug.
At some LCSC parts, it is importing the HTML-Tags:
[image]

And I have two questions, is mass import from info provider possible/planned?
Is import of part paramters support? I just tried the same fuse holder and I don't get any parameters.
For example, you order 100 different parts from mouser you just import your order xml file.