Part-DB/Part-DB-server #541: Use categories from external provider when creating parts

nomike, 2024-03-04:
**Is your feature request related to a problem? Please describe.**
When I'm looking up a part at an external provider, e.g. DigiKey, the "Create Part" mask shows the category:

```plaintext
Provider: Crystals, Oscillators, Resonators -> Crystals
```

In order to use the same category on local parts, I have to manually copy/paste the category in the field directly above.
This is mildly annoying. But the biggest issue is, that it's quite easy to forget to do that and then the part ends up in the wrong category as the last one created seems to be the one selected per default.

**Describe the solution you'd like**
I have multiple ideas on how to solve this:

1. There could be a button which, when clicked, automates the copy/paste process and enters the provider category into the category control.
2. The provider's category could be used as a default for that field, instead of just using the last one created, which is rarely the correct choice.

IMHO option 2. is the more user friendly. In most cases it does the right thing and you have to do nothin. If the provider category is not the one you want to use, you have to manually edit the category field.

Right now, you have to manually edit the category field almost every time.

Option 1 is less breaking the existing behavior, but it has the drawback that you need to press a button on every import and need to make sure not to forget it.

**Additional context**
One problem this might add is if you source parts from different providers, as they will most likely have different category names and hierarchies. So you may end up with similar resistors being in two different categories.
But that's a problem already existing right now and it doesn't change when this feature is added.

There might be ways to solve this (e.g. by having a mapping between provider categories and part-db categories. But that sounds like way more effort to implement, a lot of work for the user in maintaining that mapping between providers and is definitely out of scope for this feature request.

mkne, 2025-09-01:
> There might be ways to solve this (e.g. by having a mapping between provider categories and part-db categories. But that sounds like way more effort to implement, a lot of work for the user in maintaining that mapping between providers and is definitely out of scope for this feature request.

This is what I'm usually doing: I add the name of the information providers category as an alternative name to my part-db category (Edit category->properties->Advanced->Alternative Names). This works in many cases except if there is a comma in the category of the information provider. But it's still manual work.

Ainndy, 2026-10-01:
@jbtronics 

Hello,

I am using the information provider feature to create parts from suppliers.

Currently I use category Alternative Names to map supplier categories to my local Part-DB categories.

This works for categories without commas, but fails when the alternative name itself contains a comma.

I guess Alternative Names are internally stored or parsed as a comma-separated list, so the comma inside a category name is treated as a separator.

Could Alternative Names support escaping or another separator format?

Possible solutions:
- Store alternative names as an array instead of a comma-separated string
- Support quoted values, e.g.
  "Diodes, Rectifiers", Single Diodes
- Use semicolon as separator
- Provide a dedicated UI where every alternative name is a separate entry

This would make supplier category mapping much easier for users who use DigiKey, Mouser, LCSC, etc.

Thanks!