Part-DB/Part-DB-server #544: Database fails when "Update existing part from info provider"

BlackDrakon, 2024-03-05:
**Describe the bug**
If I do update existing part from info provider(octopart): [TPS563207SDRLR] it fails with error 500.

**Expected behavior**
I got this bug on two different parts (TPS563207SDRLR and TPS54623) many others saved fine. Other providers I can not test.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: 1.10.7 and 1.11.1

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser chrome

jbtronics, 2024-03-05:
What does the logs of Part-Db say? You should find some more detailed error message there.,

BlackDrakon, 2024-03-05:
I found a couple of similar messages like this:

> {"message":"Deprecated:` strtolower(): Passing null to parameter #1 ($string) of type string is deprecated","context":{"exception":{"class":"ErrorException","message":"Deprecated: strtolower(): Passing null to parameter #1 ($string) of type string is deprecated","code":0,"file":"/var/www/partdb/vendor/symfony/validator/Constraints/TypeValidator.php:74"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2024-02-27T10:21:42.299548+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/part/from_info_provider/octopart/68058745/create","ip":"192.168.30.189","http_method":"POST","server":"stockpit.eb.local","referrer":"http://stockpit.eb.local/en/part/from_info_provider/octopart/68058745/create"}}

and also this:

> {"message":"Uncaught PHP Exception Doctrine\\DBAL\\Exception\\DriverException: \"An exception occurred while executing a query: SQLSTATE[22001]: String data, right truncated: 1406 Data too long for column 'value_text' at row 1\" at ExceptionConverter.php line 117","context":{"exception":{"class":"Doctrine\\DBAL\\Exception\\DriverException","message":"An exception occurred while executing a query: SQLSTATE[22001]: String data, right truncated: 1406 Data too long for column 'value_text' at row 1","code":1406,"file":"/var/www/partdb/vendor/doctrine/dbal/src/Driver/API/MySQL/ExceptionConverter.php:117","previous":{"class":"Doctrine\\DBAL\\Driver\\PDO\\Exception","message":"SQLSTATE[22001]: String data, right truncated: 1406 Data too long for column 'value_text' at row 1","code":1406,"file":"/var/www/partdb/vendor/doctrine/dbal/src/Driver/PDO/Exception.php:28","previous":{"class":"PDOException","message":"SQLSTATE[22001]: String data, right truncated: 1406 Data too long for column 'value_text' at row 1","code":22001,"file":"/var/www/partdb/vendor/doctrine/dbal/src/Driver/PDO/Statement.php:130"}}}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2024-02-27T10:21:43.631434+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/part/from_info_provider/octopart/68058745/create","ip":"192.168.30.189","http_method":"POST","server":"stockpit.eb.local","referrer":"http://stockpit.eb.local/en/part/from_info_provider/octopart/68058745/create"}}

As I understood a string is too long, which one I did not get.
Should I publish the full log here?

jbtronics, 2024-03-05:
Its the "Schedule B" parameter returned by Octopart which returns 267 (pretty nonsensical looking) characters.

Im not sure whether to consider this as a bug in Part-DB (maybe that you get an 500 error instead of an validation error), or if octopart too long data in that case.

BlackDrakon, 2024-03-06:
If it would be validation error, it will not be a problem in PartDB (from my point of view). In this issue I got Error 500 as you said and no notifications through the interface what exact I should correct to avoid this error.

jbtronics, 2024-03-09:
The length now get properly validated.

closed 2024-03-09 by jbtronics