Part-DB/Part-DB-server #546: Label templates with placeholders have more macros than Twig templates?

frank-f, 2024-03-06:
**Is your feature request related to a problem? Please describe.**
I'm trying to create pretty labels for my parts. I have parts from official distributors where all fields are properly filled, so I have the MPN and everything, but I also have quite a few parts from less trustworthy 3rd party sources, where I don't have the MPN (and sometimes don't even know the manufacturer). In those cases I'd like to print the part name in place of the MPN.
My first idea was something like [[MPN_OR_NAME]], but that does not exist. I realize that that would probably be too specially fit for my case anyways. Others would then come and request more quirks like this.
So I enabled Twig mode and tried to achieve the same with it. The following works perfectly to achieve my goal:
`{% if element.mpn %}{{ element.mpn }}{% else %}{{ element.name }}{% endif %}`
Unfortunately I noticed, that Twig mode disables Placeholder processing. There I ran into the problem, that the available fields are not very well documented and I'm struggling to find the Twig names for (for example) [[CATEGORY_FULL]] and [[BARCODE_QR]].
 
**Describe the solution you'd like**
Could you maybe run the placeholder replacement on top of Twig? In my thinking that would probably be the quickest solution?

**Describe alternatives you've considered**
Please at least document the Twig variable names for all available Placeholder macros.

**Additional context**
If I switch to Twig mode, the already pretty limited WYSIWYG editor becomes completely useless (to me). I think it would make sense to disable the WYSIWYG editor for Twig templates or at least make "Source" the default mode.

jbtronics, 2024-03-06:
> Could you maybe run the placeholder replacement on top of Twig?

Im not a huge fan of this as it might mess up the output of twig and is not really flexible. But I can add a twig function, which returns the value of a placeholder for the given element (e.g. something like `{{ p(element, "[[CATEGORY_FULL]]" }}).

That way you can work with the returned values in twig (and modify it).

> There I ran into the problem, that the available fields are not very well documented and I'm struggling to find the Twig names for

Basically the twig mode has full access to the PHP element of the data with all its fields and methods. Thats why the twig mode is disabled by default, because it can allow users to access data that they might should not see according to their permissions (and maybe even change it somehow).

I would not give a backwards compatibility guarantee of the field names and its a lot of fields, so they are not really documented. But basically it should be all the fields, that are available in the API endpoints.

For the full path of a category it is for example `{{ element.category.full_path }}` or `{{ element.category.full_path(" -> ") }}` if you want another seperator.

frank-f, 2024-03-06:
> Im not a huge fan of this as it might mess up the output of twig and is not really flexible. But I can add a twig function, which returns the value of a placeholder for the given element (e.g. [...]

That sounds like something I could work with, too. I also tried your `element.category.fullPath` suggestion just now and that works fine. If I have to change the entire template anyways, the function is not really needed. 

> Basically the twig mode has full access to the PHP element of the data with all its fields and methods. Thats why the twig mode is disabled by default, because it can allow users to access data that they might should not see according to their permissions (and maybe even change it somehow).

I realize that, but I'm the only user in my instance, so I can live with that. And the security risk is the identical if I additionally run the Placeholder replacement or not, isn't it?

> I would not give a backwards compatibility guarantee of the field names and its a lot of fields, so they are not really documented. But basically it should be all the fields, that are available in [...]

That works, nice! My problem is, that I don't know these names and I'm not an expert in Symfony or Twig or any of this, so I'm having trouble finding them in the code. Also the API docs have other field names in some cases, like this one for example, which is called full_path.

Now the only thing I can't get working is `{{ barcode }}`. In the form I selected "None", because I want to use the plain template without any div containers and such. From what I can read in the code, that disables the barcode renderer, because it would not know which one to produce. Do you have any quick idea how to work around that? I'd need `[[BARCODE_QR]]`. 😅

frank-f, 2024-03-06:
In case you (or anyone else) is wondering where I'm trying to get with these requests. I want my labels to look like this:
[image]

jbtronics, 2024-03-06:
With the latest commits there is now a `placeholders` filter available in twig mode, which can replace the placeholders of a given string or using the  `{% apply %}` construct of a full block. See https://docs.part-db.de/usage/labels.html#use-the-placeholders-in-twig-mode

frank-f, 2024-03-06:
It sounds like a very nice solution. I tested it and noticed one problem however. If I try to render `[[BARCODE_QR]]` I see the img Tag instead of the image. Can I disable HTML escaping in Twig or is that something that needs to be done in the code?

closed 2024-03-06 by frank-f